zsa/wally #3: Error while rebooting the device

FredricW, 2019-05-20:
Using the Ergodox EZ Glow (through the Apple HDMI/USB dongle), I get an error when Wally tries to reboot the keyboard after the flashing process.

**Specs**
- MacBook Pro late 2018 with MacOS Mojave 10.14.4
- Wally v1.0.0

Here is a screenshot as it seems I cannot copy the code from the built in terminal.
[image]

The actual keyboard does reboot though and everything seems to work fine after the flash, but the program does not seem to get to the last stage.

lee-dohm, 2019-05-20:
I got the same error:

* 2019 27-inch Retina iMac with macOS Mojave 10.14.5
* USB 2 hub between iMac and keyboard
* ErgoDox EZ Shine
* libusb stable v1.0.22 installed via brew
* [Keyboard firmware HEX file](https://github.com/lee-dohm/keyboard-firmware/blob/master/keyboard-firmware.hex)

fdidron, 2019-05-20:
Seems like this affect users connecting through a hub or dongle. I am looking for a hub to reproduce the bug, @lee-dohm since you have libusb installed can you confirm you get the same error with the CLI ?

fdidron, 2019-05-21:
@lee-dohm Can you point me to the brand/model of your USB Hub ?

lee-dohm, 2019-05-21:
Same overall behavior, but no error message:

```
$ wally-cli ergodox-keyboard-firmware.hex
 32760 / 32256 [===========================================================================================================] 101.56%
^C⏎ 
```

Basically, it gets to the end and just hangs there. The keyboard is usable and has been updated, but the CLI never exits. I have to use <kbd>Ctrl+C</kbd> to return control to the shell.

[Brand and model of the USB hub](https://www.thinkgeek.com/product/e5be/). I'm trying to see if there's someone they OEMed it from.

skullface, 2019-06-09:
👋 Same here between rMBP (mid-2015, macOS 10.14.15) ⇄ [powered USB 2.0 hub](https://www.amazon.com/Anker-Aluminum-Portable-Adapter-Devices/dp/B00PBZX0OM?ref_=bl_dp_s_web_2528932011) via [dock’s USB connector](https://hengedocks.com/pages/vertical-macbook-pro-retina) ⇄ ErgoDox EZ using the Wally GUI.

cookandy, 2019-06-20:
Same issue here on a MBP (2017 model - 10.14.5) when using the [Apple USB-C adapter](https://www.apple.com/shop/product/MJ1K2AM/A/usb-c-digital-av-multiport-adapter?fnode=8b). Neither UI or CLI work.

ebabel, 2019-07-03:
Also not working for me on hub, but with this [adapter](https://www.amazon.ca/AUKEY-Adapter-MacBook-Chromebook-Pixelbook/dp/B01C858SU8) it does work.

fdidron, 2019-07-03:
Just got my hand on a hub that reproduces the issue, will work on a fix very soon :)

borkyborkbork1, 2019-07-25:
Same issue with Win10, no hub, USB 3.0.  Worked for initial simple flash then started showing exact same log/error as this issue.

fdidron, 2019-08-09:
Hi there sorry for it has been quite a difficult issue to reproduce and solve. Can you try this version of Wally https://github.com/zsa/wally/releases/download/1.1.1-osx/wally-osx-1.1.1.dmg

@borkyborkbork1 I'll be working on a similar fix for Windows as well.

mjalas, 2019-08-13:
@fdidron, I just tried the new version and it doesn't get stuck in the GUI anymore nor shows any errors. However, the keyboard doesn't reboot and I had to replug it for it to react to key presses. I'm also using a usb-hub on a MacBook Pro 2017, macOS Mojave 10.14.5.

JaredCorduan, 2019-09-01:
I run into the same error with:
* ThinkPad X1 Carbon (6th Gen)
* ThinkPad USB-C Dock
* Ubuntu 18.04.3
* Wally v1.1.0
* Ergodox EZ Glow

If I _do not_ use the dock then there is no error, but with the dock I get the error, though keyboard does reboot  and everything seems fine afterwards.

nsteinmetz, 2019-09-03:
Hello,

For macosx,I had the same error using the official Apple usbc to usb/hdmi/usbc adapter and/or a dell dock, till I installed `libusb`via `homebrew`:

```
brew install libusb
```

And not it works like a charm - it should be stated as a pre-requisites in the doc for OSX Install (as it is for Linux install)

rspeicher, 2019-10-11:
Just ran into this issue on a MacBook that only has USB-C ports, using this hub: https://www.amazon.com/gp/product/B071G83L1J/ref=ppx_yo_dt_b_asin_title_o04_s00?ie=UTF8&psc=1

Installing `libusb` through Homebrew and then flashing with `wally-cli` seems to have worked.

tchajed, 2019-10-12:
I'm still running into this with:
- MacBook Pro mid-2015 running macOS 10.14.6
- Ergodox EZ Glow
- AmazonBasics USB 2.0 hub (note there's no USB-C involved, it's USB to USB)
- wally-cli 1.1.3
- libusb 1.0.23 installed through Homebrew

It's not a serious issue because it seems like flashing works fine, the CLI just hangs.

Also, the progress bar is incorrect even without the hub (it ends up at 32760 / 32256 bytes, 101.56%), although unlike with the hub it terminates successfully.

finalgriever, 2019-12-13:
Just to contribute to the error profile, I'm getting the same issue on the Microsoft Surface Book 2.

timolehto, 2020-02-14:
I ran into this issue as well. I don't remember having any issues 4 months ago when I last updated my firmware, but now the reboot fails.

astrelochnik, 2020-02-18:
same issue on a dual boot Arch Linux latest release and on Windows 10 Ed. I get it in both OS's.
Thinkpad t580. wally v1.1.4. Ergodox EZ.

Despite the error. It still flashes the new config onto the keyboard.

Jill1627, 2020-03-11:
Same issue with Macbook and MBP with Apple HDMI dongle. 
The only dongle that worked for me is a  Tripp-Lite one.

jaredpar, 2020-03-11:
Same issue with a Windows Surface Laptop connected through a docking station. Was able to work around the problem by unplugging the keyboard and plugging it back in. Everything worked fine after that.

JesusTheHun, 2020-03-13:
Same issue with a MBP and a combo Usb hub + usb toggler.

sajjad, 2020-05-02:
Similar issue with a Dell XPS 15 9560 on Win 10. Wally doesn't detect keyboard when connected through a USB hub.

bsansouci, 2020-05-15:
Same issue here on MBP with a dongle!

qhkm, 2020-05-22:
Same issue here on MBP with a donggle

sinewalker, 2020-06-04:
@fdidron the updated `dmg` works for me over a [mbeat usb 3 hub](https://djcity.com.au/product/mbeat-usb-m43hub-7-port-usb-3-0-and-usb-2-0-hub-manager-with-switch/) connected via a usb-c to usb-a dongle from a 2019 MBP

before trying the new image i tested via only the dongle, which also worked. i already had `libusb` installed from homebrew too, for a different reason.

bedge, 2020-06-06:
Same issue with mid 2015 macbook pro, mojave 10.14.6
Using hub in HP HC270cr monitor

bedge, 2020-06-07:
This may be more help, the dtrace output of a full run, not just the bit after it hangs like I attached above.
[wally-dtrace.txt](https://github.com/zsa/wally/files/4741208/wally-dtrace.txt)

arcaartem, 2020-06-15:
Same issue on MBP 16, connected through a USB hub (Product ID: 0x0610, Vendor ID: 0x05e3). Wally v1.1.3

fdidron, 2020-06-16:
Hi everyone, apologies for the lack of updates on this.
Once I'm done [with this](https://github.com/zsa/wally/issues/34#issuecomment-644541798) I will look into fixing this issue.
The idea is to ditch libusb for Windows and OS X and use native code for usb communication. I already have a proof of concept on Windows, but for OS X I'm entering new code language territory.

lawfuller, 2020-09-09:
I get the same error on Windows 10.0.19041 Build 19041. The keyboard's driver info is attached.
[image]

xxxserxxx, 2020-09-28:
Just FYI, in case it helps at all -- it's not just dongles. I have an XPS connected to a Kensington USB-C dock (over USB-C) that behaves the same. The dock's providing 2x external monitors, power, and all peripherals over USB-C.  There's no DisplayLink in the mix - it's all native.  The dock is not Thunderbolt compliant, so I can't test whether it behaves the same over TB.  

The laptop is running Arch Linux and the 5.8.10 kernel.

vitvakatu, 2020-11-05:
same issue here on MacOS Catalina. @fdidron do you need any help with this?

fdidron, 2020-11-25:
@xxxserxxx @lawfuller @jaredpar @astrelochnik @thestillheron @borkyborkbork1 
There's a new version of Wally for Windows here https://github.com/zsa/wally-win . Rewritten using native technologies, it should fix that kind of issues. Feel free to open a new issue in the new repo if needed.

I'm starting the same rewrite for OS X (Including support for M1 Chips).

fdidron, 2020-11-25:
A quick question to everyone using Mac OS subbed to this thread, are you using OS X Catalina / Big Sur ?

I'm asking because Apple deprecated accessing usb devices via IOKit starting from Catalina, and if I choose to use the recommended way, Wally's rewrite will only be running on Catalina+ releases.

arcaartem, 2020-11-25:
I'm on Catalina at the moment and probably will continue hold off upgrading to Big Sur for a while.

lawfuller, 2020-11-25:
Thanks for working on this!

 

BTW, you may want to state somewhere that this requires a system restart. Before I restarted Windows, the updated Wally didn’t work. When I tried to flash a new bin, I got error 5. Sometimes I got error 121.

 

 

11:50:44 - Error - Device handle no good. Error code: 5

11:50:44 - Info - Target stm32 found, the keyboard is in reset mode.

11:50:44 - Info - Starting flash process, targeting stm32 devices.

11:50:44 - Info - Firmware file is valid for stm32.

11:50:44 - Info - Selected firmware, path: C:\Users\lawfu\OneDrive\Desktop\Moonlander Layouts\moonlander_thinqu-home-row_3yqRw_4aa63.bin

11:50:43 - Info - Device selected as a flashing target: Keyboard in Reset Mode | stm32

11:50:43 - Info - 1 - Keyboard in Reset Mode | \\?\usb#vid_0483 <file://%3f/usb%23vid_0483&pid_df11%232056336a2033%23%7ba5dcbf10-6530-11d2-901f-00c04fb951ed%7d> &pid_df11#2056336a2033#{a5dcbf10-6530-11d2-901f-00c04fb951ed} | stm32.

11:50:43 - Info - Enumeration complete for target, found 1 devices.

11:50:42 - Info - Starting enumeration, targeting any compatible devices.

11:50:42 - Info - Restarting the flash process.

11:50:40 - Error - Device handle no good. Error code: 5

11:50:40 - Info - Target stm32 found, the keyboard is in reset mode.

11:50:40 - Info - Starting flash process, targeting stm32 devices.

11:50:40 - Info - Firmware file is valid for stm32.

11:50:40 - Info - Selected firmware, path: C:\Users\lawfu\OneDrive\Desktop\Moonlander Layouts\moonlander_thinqu-home-row_3yqRw_m9moP.bin

11:50:38 - Info - Device selected as a flashing target: Keyboard in Reset Mode | stm32

11:50:38 - Info - 1 - Keyboard in Reset Mode | \\?\usb#vid_0483 <file://%3f/usb%23vid_0483&pid_df11%232056336a2033%23%7ba5dcbf10-6530-11d2-901f-00c04fb951ed%7d> &pid_df11#2056336a2033#{a5dcbf10-6530-11d2-901f-00c04fb951ed} | stm32.

11:50:38 - Info - Enumeration complete for target, found 1 devices.

11:50:38 - Info - Starting enumeration, targeting any compatible devices.

11:50:38 - Info - Restarting the flash process.

11:50:33 - Error - Device handle no good. Error code: 5

11:50:33 - Info - Target stm32 found, the keyboard is in reset mode.

11:50:33 - Info - Starting flash process, targeting stm32 devices.

11:50:33 - Info - Firmware file is valid for stm32.

11:50:33 - Info - Selected firmware, path: C:\Users\lawfu\OneDrive\Desktop\Moonlander Layouts\moonlander_thinqu-home-row_3yqRw_4aa63.bin

11:47:47 - Info - Device selected as a flashing target: Keyboard in Reset Mode | stm32

11:47:47 - Info - 1 - Keyboard in Reset Mode | \\?\usb#vid_0483 <file://%3f/usb%23vid_0483&pid_df11%232056336a2033%23%7ba5dcbf10-6530-11d2-901f-00c04fb951ed%7d> &pid_df11#2056336a2033#{a5dcbf10-6530-11d2-901f-00c04fb951ed} | stm32.

11:47:47 - Info - Enumeration complete for target, found 1 devices.

11:47:47 - Info - Starting enumeration, targeting any compatible devices.

11:47:47 - Info - Restarting the flash process.

[... 66 lines cut ...]
11:45:46 - Info - Flashing: Erasing flash.

11:45:46 - Info - Flashing: Initializing.

11:45:46 - Info - Target stm32 found, the keyboard is in reset mode.

11:45:42 - Info - Starting flash process, targeting stm32 devices.

11:45:42 - Info - Firmware file is valid for stm32.

11:45:42 - Info - Selected firmware, path: C:\Users\lawfu\OneDrive\Desktop\Moonlander Layouts\moonlander_thinqu-home-row_3yqRw_m9moP.bin

11:24:05 - Info - Wally is up to date.

11:24:04 - Info - Checking for updates.

11:23:59 - Info - Device selected as a flashing target: Moonlander MK1 | stm32

11:23:59 - Info - 1 - Moonlander MK1 | \\?\hid#vid_3297 <file://%3f/hid%23vid_3297&pid_1969&mi_01&col01%238&2a33b138&0&0000%23%7b4d1e55b2-f16f-11cf-88cb-001111000030%7d> &pid_1969&mi_01&col01#8&2a33b138&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030} | stm32.

11:23:59 - Info - Enumeration complete for target, found 1 devices.

11:23:59 - Info - Starting enumeration, targeting any compatible devices.

11:23:59 - Info - Application started

sklum, 2020-11-30:
> A quick question to everyone using Mac OS subbed to this thread, are you using OS X Catalina / Big Sur ?

I'm on Big Sur on an M1, can't get the Moonlander to connect at all.  Happy to help with testing.

TheNerdyHamster, 2020-12-10:
I do have the same problem on Archlinux with wally 2.1.0.
But when I unplugged the keyboard and plugged it in agian. everything worked fine.

xxxserxxx, 2020-12-11:
**Wally**: *2.1.0-10*  
**Installed from**: *Arch AUR*  
**OS**: *Linux 5.9.12-arch1-1*  
**KB**: *2017 Ergodox EZ Shine*

It doesn't hang anymore.  I see only two issues, which I don't know are related:

1. Now the transfer progress bar doesn't complete; it's always short by a couple Kb.  However, I haven't yet detected any bindings that don't work, so I've been assuming it's just a UI issue.
2. The glow is *always* wrong. The left keyboard colors are correct, but the right board's underglow is always blue.  This is unless I unplug the keyboard for a dozen seconds -- when it comes back up the colors are correct, for a while.  This is **not** new.  The shine has been this way for months, since long before the update.

This version is a net improvement for me, since wally-cli does not hang any longer, the key mappings seem to be correct.

fdidron, 2020-12-16:
Hello, some progress on Wally's rewrite for OSX

We have been testing a cli version internally and you can now find it here:
https://github.com/zsa/wally/releases/tag/3.0.0osx-beta

To use it, open a terminal and go to the folder where you downloaded wally-cli. Then run `./wally-cli <path to your .bin file>`

If anyone who had the issue of the keyboard not rebooting and report could give it a try, that would be great.

fdidron, 2020-12-16:
I'm starting work on the GUI now

arcaartem, 2020-12-16:
@fdidron I've just gave it a go. Failed with the following error message:
```
⠼ Flashing firmware do not unplug your keyboard.255
255
Something went terribly wrong.
The operation couldn’t be completed. (UsbLibrary.DFUError error 0.)
```

This is the command line that I used:
```
./wally-cli ./ergodox_ez_shine_the-red-nomad-v5-2_l7ZQ0_brme7.hex
```

I have recently upgraded to Big Sur

xxxserxxx, 2020-12-16:
@fdidron -- some of us have this issue, but on Linux. Is this version intended to only address issues on OSX?  Is there any reason for non-Mac users to give it a try, aside from testing possible cross-platform regressions?

danielhoherd, 2021-03-16:
I just experienced this on macOS 11.2.3 with Wally 2.1.0 and a [Lention](https://www.amazon.com/gp/product/B06X1G6FWX/) USB hub. Switching to a straight USB-A to USB-C adapter solved the issue.