zsa/wally #5: Add to Brew Cask?

joshmedeski, 2019-05-21:
I would love if you guys could add this to Brew cask!

drashna, 2019-06-13:
@fdidron we would need a `Wally.app.zip` to be able to do this. 

And for reference, this is the QMK Toolbox:
https://github.com/Homebrew/homebrew-cask-drivers/blob/master/Casks/qmk-toolbox.rb

However, this should be simple enough to do, but would allow me to more easily autoinstall wally in QMK 

Also, in case it helps: https://github.com/qmk/qmk_toolbox/blob/master/osx/build_release.command

fdidron, 2019-08-28:
Hello,
The application is available on brew cask:

brew tap homebrew/cask-drivers
brew cask install zsa-wally

closed 2019-08-28 by fdidron

shivekkhurana, 2021-01-01:
I had to run: `brew install --cask zsa-wally` instead. Why does this not show up in the brew index?

joshmedeski, 2021-01-06:
@shivekkhurana you have to tap the cask-drivers first. Run both of the commands below to install Wally with brew:

```sh
brew tap homebrew/cask-drivers
```

```sh
brew install --cask zsa-wally
```