zsa/wally #10: Window clips and prevents resize

daveriedstra, 2019-06-07:
The Wally window clips some content and won't allow me to resize it:

[image]

Pop!_OS 19.04 using wayland gnome session

Wally looks great, btw!

fdidron, 2019-06-10:
@daveriedstra I have to say I didn't try running wally on Pop!_OS, I'l spin up a VM and look for a fix.

fdidron, 2019-07-02:
@daveriedstra A new release of Wally is out and should fix clipping issues: https://github.com/zsa/wally/releases/download/1.1.0-linux/wally

daveriedstra, 2019-07-05:
@fdidron thanks, still clipping a bit but at least the buttons are visible :)

[image]

fdidron, 2019-07-09:
@daveriedstra , Oh no ! I increased some of the space, can you try this version:
https://github.com/zsa/wally/releases/download/1.1.0-linux/wally

daveriedstra, 2019-07-11:
Beautiful, thank you!

[image]

closed 2019-07-22 by fdidron