zsa/wally #9: CLI and GUI dependencies

daveriedstra, 2019-06-07:
The wiki for the linux install makes it seem like the gtk3 and webkit dependencies are also required for wally-cli. Regardless of whether this is the case or not, it should probably be documented, given the size of these dependencies (~373mb on system).

fdidron, 2019-06-10:
Very good point @daveriedstra, thanks. I updated the wiki accordingly

closed 2019-06-10 by fdidron

zanona, 2019-12-12:
Hey, @fdidron how can I run the cli?
I downloaded the bin file and it keeps asking for dependencies:

> ./wally: error while loading shared libraries: libwebkit2gtk-4.0.so.37: cannot open shared object file: No such file or directory

zanona, 2019-12-12:
@fdidron nevermind, I just found the wally-cli url by the checking the AUR package sources Perhaps it could also be added to the wiki when providing the download links? 
https://github.com/zsa/wally/releases/download/1.1.1-linux/wally-cli