zsa/wally #12: Unable to flash firmware, OpenDevices error.

steeef, 2019-06-19:
Using a Planck EZ on a MacBook Pro Mid-2015.
When I open up Wally, I get these messages in the console:

[image]

When I select firmware and then hit the reset button on my keyboard, nothing happens.

Running `wally-cli` gives me similar results. I get the prompt to press the reset button, I do, and nothing happens.

I've tried rebooting, plugging directly into the MacBook (normally I use a hub), and swapping out the cable. No dice.

Weirdly enough, this worked yesterday multiple times, without issue.

steeef, 2019-06-19:
Tried a few other things. I noticed that the error message occurs whether or not my Planck is plugged in, so it may be a red herring.

steeef, 2019-06-19:
Unplugged everything and just plugged in the Planck and it worked. Didn't narrow down which device was causing the issue, but whatever generated that error above must've prevented Wally from being able to send firmware.

fdidron, 2019-06-20:
@steeef would you mind sharing the h/w plugged into your Macbook when this happened ?

steeef, 2019-06-20:
I narrowed it down to a USB DAC and amplifier that I use. Screenshot of the info from System Information: 
[image]

As long as that device is unplugged, firmware update works as usual.

fdidron, 2019-06-22:
@steeef this is useful thanks. This error is more of a warning than an error, I will have Wally ignore it so you can flash your Planck without the need to unplug your DAC

vbabiy, 2019-07-01:
I have the same problem when I unplug the DAC works fine.

fdidron, 2019-07-02:
Hello I release  new version of Wally that should allow you to flash with the DAC plugged in:

https://github.com/zsa/wally/releases/tag/1.1.0-osx

vbabiy, 2019-07-02:
works great for me. Thanks @fdidron

steeef, 2019-07-02:
Yup, works for me too. Thanks @fdidron!

closed 2019-07-02 by steeef

dfontana, 2019-07-04:
Hi @fdidron, I ran into the same issue posted here but on Linux. Unplugging my DAC/Amp did the trick as noted by other kind posters. I tried the latest release for linux (1.1.0) but it doesn't look like the fix you introduced worked in this case. While its not a hassle by any stretch, thought I'd share

Error in Wally:

  ```
  OpenDevices: config 1 interface at index 2 has number 3, USB standard states they should be identical
  ```

I poked around `dmesg` to fetch some information that looked helpful, not sure though.

<details>
  <summary>Ergodox Info</summary>

  ```
  [13817.623846] usb 5-2: new full-speed USB device number 10 using xhci_hcd
  [13817.956247] usb 5-2: New USB device found, idVendor=feed, idProduct=1307, bcdDevice= 0.01
  [13817.956250] usb 5-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
  [13817.956252] usb 5-2: Product: ErgoDox EZ
  [13817.956253] usb 5-2: Manufacturer: ErgoDox EZ
  [13817.956255] usb 5-2: SerialNumber: 0
  [13817.974549] input: ErgoDox EZ ErgoDox EZ as /devices/pci0000:00/0000:00:1d.2/0000:06:00.0/usb5/5-2/5-2:1.0/0003:FEED:1307.000E/input/input36
  [13818.036562] hid-generic 0003:FEED:1307.000E: input,hidraw0: USB HID v1.11 Keyboard [ErgoDox EZ ErgoDox EZ] on
  usb-0000:06:00.0-2/input0
  [13818.057870] input: ErgoDox EZ ErgoDox EZ System Control as /devices/pci0000:00/0000:00:1d.2/0000:06:00.0/usb5/5-2/5-2:1.1/0003:FEED:1307.000F/input/input37
  [13818.116128] input: ErgoDox EZ ErgoDox EZ Consumer Control as /devices/pci0000:00/0000:00:1d.2/0000:06:00.0/usb5/5-2/5-2:1.1/0003:FEED:1307.000F/input/input38
  [13818.116237] input: ErgoDox EZ ErgoDox EZ Keyboard as /devices/pci0000:00/0000:00:1d.2/0000:06:00.0/usb5/5-2/5-2:1.1/0003:FEED:1307.000F/input/input39
  [13818.116561] hid-generic 0003:FEED:1307.000F: input,hidraw4: USB HID v1.11 Keyboard [ErgoDox EZ ErgoDox EZ] on
  usb-0000:06:00.0-2/input1
  ```
</details>

<details>
  <summary>JDS Labs DAC/AMP Info</summary>

  ```
  [13827.319966] usb 1-6: new full-speed USB device number 5 using xhci_hcd
  [13827.510787] usb 1-6: config 1 has an invalid interface number: 3 but max is 2
  [13827.510789] usb 1-6: config 1 has an invalid interface number: 3 but max is 2
  [13827.510790] usb 1-6: config 1 has an invalid interface number: 3 but max is 2
  [13827.510791] usb 1-6: config 1 has an invalid interface number: 3 but max is 2
  [13827.510792] usb 1-6: config 1 has no interface number 2
  [13827.526831] usb 1-6: New USB device found, idVendor=262a, idProduct=1048, bcdDevice= 0.01
  [13827.526833] usb 1-6: New USB device strings: Mfr=1, Product=2, SerialNumber=0
  [13827.526834] usb 1-6: Product: JDS Labs Element DAC
  [13827.526836] usb 1-6: Manufacturer: Yoyodyne Consulting
  [13827.541503] input: Yoyodyne Consulting JDS Labs Element DAC as /devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.0/0003:262A:1048.0010/input/input40
  [13827.600228] hid-generic 0003:262A:1048.0010: input,hidraw5: USB HID v1.00 Device [Yoyodyne Consulting JDS Labs Element DAC] on usb-0000:00:14.0-6/input0
  ```
</details>

<details>
  <summary>OS Info</summary>

  ```
  ➜ screenfetch
                            ./+o+-       
                    yyyyy- -yyyyyy+     OS: Ubuntu 19.04 disco
                ://+//////-yyyyyyo      Kernel: x86_64 Linux 5.0.0-15-generic
            .++ .:/++++++/-.+sss/`      Uptime: 4h 21m
          .:++o:  /++++++++/:--:/-      Packages: 2295
          o:+o+:++.`..```.-/oo+++++/    Shell: zsh 5.5.1
        .:+o:+o/.          `+sssoo+/    Resolution: 3840x1200
    .++/+:+oo+o:`             /sssooo.  DE: GNOME
  /+++//+:`oo+o               /::--:.   WM: GNOME Shell
  \+/+o+++`o++o               ++////.   WM Theme: Pop
    .++.o+++oo+:`             /dddhhh.  GTK Theme: Pop-slim-dark [GTK2/3]
        .+.o+oo:.          `oddhhhh+    Icon Theme: Pop
          \+.++o+o``-````.:ohdhhhhh+    Font: Fira Sans Bold 10
          `:o+++ `ohhhhhhhhyo++os:      CPU: Intel Core i7-6700K @ 8x 4.2GHz [27.8°C]
            .o:`.syhhhhhhh/.oo++o`      GPU: GeForce RTX 2080
                /osyyyyyyo++ooo+++/     RAM: 4085MiB / 15974MiB
                    ````` +oo+++o\:
                            `oo++.
  ```
</details>

fdidron, 2019-09-25:
@dfontana So sorry I only saw this now😱, can you reopen the issue or create a new one next time ?

Can you try version 1.1.1 of Wally available here https://github.com/zsa/wally/releases/tag/1.1.1-linux