zsa/wally #13: [Mac] High DPI support

noahtallen, 2019-06-25:
Hey there! I just tried Wally on my computer. I'm using a 4k monitor, and the UI looks extremely pixelated. It even shows up when you look closely at this screenshot:

[image]

It'd be really nice if it behaved nicely with high-dpi resolutions. Do you know where this might be in the code?

fdidron, 2019-07-02:
Hi there, can you try this version: https://github.com/zsa/wally/releases/tag/1.1.0-osx

My MacBookAir is unfortunately too old to connect to my 4k screen, so I cannot test it myself.

noahtallen, 2019-07-02:
I’m getting this message when I try to open Wally from the disk image:

> “Wally” is damaged and can’t be opened. You should eject the disk image.

fdidron, 2019-07-03:
@noahtallen I messed up the release, just reuploaded a version, can you give it a try ?

noahtallen, 2019-07-03:
Just tried it out, and it looks beautiful on my 4k monitor! Thanks for the fix :)

closed 2019-07-03 by noahtallen