zsa/wally #14: Add to Chocolatey

zored, 2019-06-26:
Related to #5 
Windows Choco package will be perfect too!
I have a [working example](https://github.com/zored/choco-specs/tree/master/igdm) of package.

digitalcoyote, 2019-10-19:
I'd be willing to add a package for this to my AU repo (Automatically Updates the package as Windows releases are detected). It should require no changes to your release process, and I can add this projects maintainer(s) as maintainers on the package to add credibility and give them management rights.

fdidron, 2020-11-25:
@zored 
There's a new version of Wally for Windows here https://github.com/zsa/wally-win . Rewritten using native technologies. We are accepting PRs for alternative distribution channels like Chocolatey in the new repo