zsa/wally #18: Error trying to flash Planck

Skattered, 2019-07-18:
[image]

I get this error every time I try to flash my Planck from my laptop. I have tried with the usb-a adapter and with USB-c, and I have reinstalled the program and rebooted a couple times. 

I tried on another computer and wally crashes everytime (just generic not responding crash, no info displayed to me) I click 'Select File' whether I select my ergodox or my planck.

voider1, 2019-07-18:
Having the same problem over here.

fdidron, 2019-07-19:
Hi,
Can you try to install that new version 

https://github.com/zsa/wally/releases/download/1.1.2-win/wally-winx64-1.1.2.exe

I made some changes on the Planck's bootloader.

Skattered, 2019-07-19:
I still have the same issue with 1.1.2

fdidron, 2019-07-19:
@Skattered Can you put the Planck in reset mode (pressing the reset button at the back of the keyboard with a paperclip).
Then open the Device Manager. Under "Universal Serial Bus" devices. there should be an entry called "STM32 Bootloader". 
Right click on the entry and click "Properties"
Go to the "Driver" tab then click "Driver Details".
Can you attach a screenshot of the Driver details windows ?

Thanks

Skattered, 2019-07-19:
I don't have a section called "Universal Serial Bus" just controllers and devices, but neither of those have that device listed under them. I see a libusb-win32 devices => Planck DFU Bootloader is that supposed to also be there or is it showing up different?

Skattered, 2019-07-19:
I uninstalled that driver and plugged it back in and it grabbed the driver you were talking about, and then it flashed fine.

fdidron, 2019-07-19:
@Skattered That's great, glad it worked !
Thanks for reporting.

@voider1 
Can you try the instructions above and let me know if it worked for you:

- Put the Planck in reset mode (pressing the reset button at the back of the keyboard with a paperclip).
Open the Device Manager. Under "Universal Serial Bus" devices, if there's a 
Right click on the entry and click "Properties"
Go to the "Driver" tab then click "Driver Details".
Can you attach a screenshot of the Driver details windows ?
- Then open the Device Manager. If there's an entry called libusb-win32 devices", right click on it and click "Uninstall device"
- Install Wally 1.1.2
- After rebooting ertry the flashing process.

closed 2019-07-22 by fdidron