zsa/wally #19: Not working on NixOS

waiting-for-dev, 2019-07-30:
I downloaded the binary and tried to execute it in my NixOS operating system. It simply does nothing. I can't debug anything because there is no output from the command. The only thing I can say is that the exit code (`echo $?`) is 127.

I'm trying to run the cli only version (but I tried with both `wally-cli` and `wally` binaries). In order to setup the system, I added `libusb` to `environment.systemPackages` and `udev` rules in `services.udev.extraRules` in my `configuration.nix`.

Any help is appreciated. I'm using the ErgodoxEZ.

t-hart, 2019-08-09:
I'm in the same boat. I've taken the same steps as you and with the exact same results. I think the next step would be to clone the repo and try and develop it locally.

fdidron, 2019-08-10:
Hello, I will be looking into this sometimes next week, I m currently setting up a Nix OS VM.

waiting-for-dev, 2019-08-10:
Thank you very much! Please, tell me anything you may need from me.

fdidron, 2019-08-13:
Hi @waiting-for-dev @t-hart 

Some progress here, (NixOs is quite different from other Linux distros !!!)
Can you create a file `wally-cli.nix` and paste this inside:

```
let
   pkgs = import <nixpkgs> {};
in pkgs.stdenv.mkDerivation {
  name = "wally-cli-1.1.0";
  src = pkgs.fetchurl {
      name = "wally-cli";
      url = "https://github.com/zsa/wally/releases/download/1.1.0-linux/wally-cli-uncompressed";
      sha256 = "1prsn3fqi44iwsiyh9nhs82hbd1afw98b0x0c0w248z50dddpr17";
  }; 
  dontStrip = true;
  unpackPhase = ''
    cp $src ./wally-cli
  '';
  installPhase = ''
    mkdir -p $out/bin
    chmod +wx wally-cli
    cp wally-cli $out/bin
    ${pkgs.patchelf}/bin/patchelf \
      --interpreter "$(cat $NIX_CC/nix-support/dynamic-linker)" \
      --set-rpath "${pkgs.lib.makeLibraryPath [ pkgs.libusb1 ]}" \
      $out/bin/wally-cli
  '';
}
```

Then run `nix-env -i -f wally-cli.nix` to install wally-cli .

I'm still checking on how to make it work for the gui.

t-hart, 2019-08-14:
Nice one! Yeah, this works perfectly! For me at least, a CLI is plenty, though I'm sure some would appreciate a GUI too.

fdidron, 2019-08-15:
@t-hart I am going to release recipes for both the cli and the gui in the coming days.

waiting-for-dev, 2019-08-15:
Working perfectly well here, too. Thank you very much for your chick work on this!

winpat, 2019-09-03:
@fdidron Funny, I just started packaging wally for NixOS and then I stumbled over this issue.

Let me know if there something I can do to help (testing... etc.)

davidak, 2020-05-26:
@fdidron first, thanks and respect for the effort and dedication to look into NixOS! It has a steep learning curve and is hard to learn for new users. (we try to improve that)

Have you looked into packaging the GUI?

@spacekookie added the CLI in https://github.com/NixOS/nixpkgs/commit/4e78f846b649c5743c8725586866f5506bf69bc1#diff-82c4f613221a270650ed55ccb8ea95ae.

You can now use it with Nix without installing it:

`nix run nixpkgs.wally-cli -c wally '/home/davidak/Downloads/ergodox_ez.hex'`

I thought the GUI might only need some additional dependencies and want to look into it, but dont waste time when you already have something.

Sir4ur0n, 2020-12-17:
Update: @davidak kindly opened https://github.com/NixOS/nixpkgs/pull/91203 a while ago to automatically handle the `plugdev` group creation as well as configuring the udev rules.

I tested it today for my Moonlander and it works well (with an updated Wally Git commit).

Once this is merged, all that's left to do is add in Wally documentation that for NixOS, the steps are:
* Add `hardware.keyboard.zsa.enable = true;` in `configuration.nix`
* Add the user to the `plugdev` group: `users.users.<userName>.extraGroups = [ "plugdev" ];`
* Add `wally-cli` to packages (`configuration.nix` or Home Manager `hone.nix`)

:tada:

thomasheartman, 2020-12-17:
@Sir4ur0n That's great! And thanks for the heads-up on the config change required.

But I'm not sure I follow, so out of curiosity: didn't it work previously with the `wally-cli` that's in nixpkgs? Does the Moonlander require this or is it because of the Chrome connector? Or does it relate to the GUI version of Wally?

I got the Moonlander a little while back myself and can't remember having any issues with it. However, it *is* quite possible that I've only configured it on macOS and that I don't know whether it works on NixOS or not :sweat_smile:

Sir4ur0n, 2020-12-17:
> But I'm not sure I follow, so out of curiosity: didn't it work previously with the wally-cli that's in nixpkgs?

It would be great if others give it a try, but as for me, flashing didn't work on my Moonlander Mk1. When flashing, my keyboard bipped, but the CLI command remained silent (as if still waiting for the reset button to be pressed) and the keyboard no longer worked (required unplugging and replugging the keyboard).

Also, the Chrome connector didn't work: no device was detected on https://configure.ergodox-ez.com/train

> Or does it relate to the GUI version of Wally?

No, this PR is unrelated to GUI Wally, which AFAIU still doesn't work on NixOS.

Mohammer5, 2020-12-17:
> It would be great if others give it a try, but as for me, flashing didn't work on my Moonlander Mk1. When flashing, my keyboard bipped, but the CLI command remained silent (as if still waiting for [...]

Same for me. That's happening because the moonlander requires wally-cli to be at least v2.1.0 while the version in nixpkgs is 2.0.0. The newer version requires some updates, adjusting the version and the hash didn't do the trick for me.

Sir4ur0n, 2020-12-18:
With the aforementioned PR and Wally 2.0.0 flashing works (though I still get some errors/warnings), maybe 2.1.0 removes the errors though?

thomasheartman, 2020-12-21:
@Sir4ur0n Right, thanks for the clarification! In that case it's probably that I just haven't tried it flashing it on NixOS yet. Thanks again for the update and the instructions, though; I'll remember these next time I need to  give it a go :smile:

davidak, 2020-12-26:
Yes, flashing needs the udev rules to work.

Sir4ur0n, 2021-01-07:
Can someone with correct rights update the [Linux wiki page](https://github.com/zsa/wally/wiki/Linux-install) to reflect the needed steps for NixOS please? Then I think we can close this issue?

2.1. NixOS
Add the following to your `configuration.nix` to create the group and udev rules:
```nix
users.users.<yourUser> = {
  extraGroups = [
    "plugdev"
  ];
};
hardware.keyboard.zsa.enable = true;
```

Again, thank you to @davidak :bow:

miniBill, 2021-03-01:
Adapted from @fdidron answer, the latest wally can be got with:

```let
   pkgs = import <nixpkgs> {};
in pkgs.stdenv.mkDerivation {
  name = "wally-2.1.0";
  src = pkgs.fetchurl {
      name = "wally";
      url = "https://github.com/zsa/wally/releases/download/2.1.0-linux/wally";
      sha256 = "11y0jvrm36rkai768hx0ds4an3af3rnp99ygvznq2fjfkn4nicfd";
  }; 
  dontStrip = true;
  unpackPhase = ''
    cp $src ./wally
  '';
  installPhase = ''
    mkdir -p $out/bin
    chmod +wx wally
    cp wally $out/bin
    ${pkgs.patchelf}/bin/patchelf \
      --interpreter "$(cat $NIX_CC/nix-support/dynamic-linker)" \
      --set-rpath "${pkgs.lib.makeLibraryPath [ pkgs.libusb1 pkgs.webkitgtk pkgs.gtk3 pkgs.glib ]}" \
      $out/bin/wally
  '';
}
```

thomasheartman, 2021-03-05:
Is the `hardware.keyboard.zsa.enable` really necessary? I managed to flash my moonlander without it just now. With the `plugdev` group and `wally-cli` from unstable, everything appears to work as expected. I've been putting it off because I thought I'd need to upgrade to 21.05 for it to work on stable (because the `hardware.keyboard.zsa.enable` option is missing). Have I misunderstood something?

davidak, 2021-03-06:
I needed the udev rules, so my Ergodox was recognized by the flasher. I don't know if it's different with the Moonlander.

I don't think you misunderstood something, it was just different in your case. When it was flashed successful, good for you!

thomasheartman, 2021-03-07:
On closer inspection, it could be because I've got this in my config? I remember setting that up previously, so if it's the same thing that happens when you use the `hardware.keyboard.zsa.enable` option, then maybe I *did* need it after all. Thanks!

 ```nix
services.udev.extraRules = ''
    # Teensy rules for the Ergodox EZ Original / Shine / Glow
    ATTRS{idVendor}=="16c0", ATTRS{idProduct}=="04[789B]?", ENV{ID_MM_DEVICE_IGNORE}="1"
    ATTRS{idVendor}=="16c0", ATTRS{idProduct}=="04[789A]?", ENV{MTP_NO_PROBE}="1"
    SUBSYSTEMS=="usb", ATTRS{idVendor}=="16c0", ATTRS{idProduct}=="04[789ABCD]?", MODE:="0666"
    KERNEL=="ttyACM*", ATTRS{idVendor}=="16c0", ATTRS{idProduct}=="04[789B]?", MODE:="0666"

    # STM32 rules for the Planck EZ Standard / Glow
    SUBSYSTEMS=="usb", ATTRS{idVendor}=="0483", ATTRS{idProduct}=="df11", \
        MODE:="0666", \
        SYMLINK+="stm32_dfu"
'';
```

davidak, 2021-03-07:
Yeah, it's the same thing.