zsa/wally #20: Unable to flash firmware, OpenDevices error.

ManuZiD, 2019-08-09:
Hi

It's related to this issue: https://github.com/zsa/wally/issues/12

I try it on linux but get exactly the same error:

```
22:49:28 info Application UI / State initialized
22:49:30 info Probing compatible usb devices
22:49:30 error OpenDevices: config 1 interface at index 0 has number 12, USB standard states they should be identical
22:49:30 info Found 1 compatible device(s)
```

fdidron, 2019-08-15:
Hi @ManuZiD , can you try release 1.1.1 : https://github.com/zsa/wally/releases/tag/1.1.1-linux

ManuZiD, 2019-08-15:
Hi @fdidron unfortunately I currently haven't access to my computer with linux on.

ZaneA, 2019-08-16:
I can confirm that release 1.1.1 fixes this issue for me, thanks! :)

fdidron, 2019-08-17:
@ManuZiD Closing this for now, feel free to reopen if the issue persists once you get to test the release.

ManuZiD, 2019-08-29:
Hi @fdidron i got the same error...

snitko, 2020-10-08:
I've just experienced this issue after downloading wally v2.1.0. Yesterday it worked just fine. Now, granted, it was a different PC (same Ubuntu though), but I moved to a freshly installed Ubuntu 20.04 on a new PC and downloaded the latest version of Wally. When attempting to run it, the console output is this:

```
17:36:36 info Found 1 compatible device(s)
17:36:36 warning OpenDevices: config 1 interface at index 1 has number 2, 
USB standard states they should be identical
17:36:36 info Probing compatible usb devices
```

Attempting to install the firmware results in more errors identical to this. I unplugged all USB devices except for the keyboard - no luck.

One thing I suspect it might be is the additional USB extension  PCI-e card I have in my PC. I tried connecting the keyboard to it too, but I'm getting the same error. Any ideas?

fdidron, 2020-10-08:
Hi @snitko can you confirm at which step do you get in Wally ? Do you get to the step where it asks you for a firmware ?

snitko, 2020-10-08:
Yes, I get to that step where it asks for a firmware. However, I get that error even before that step if I open console - it's there from the start. And if I proceed and select the file, it would just endlessly print that error without updating the keyboard.

fdidron, 2020-10-08:
Got it, trying to confirm things steps by steps, once you choose the firmware and pressed the keyboard's reset button, it still loops to that error ?
Also can you paste in the result of `lsusb` after pressing the keyboard's reset button

snitko, 2020-10-08:
Alright, so I checked everything in the following order:

1. Two keyboards connected (Ergodox and a regular one - I tried with Ergodox only, but it stops working after I press reset, so I needed another one to enter `lsusb` command)

2. run 'wally`

3. On step on, I go to the console an see the above mentioned error message. Then go back to the GUI.

4. It allows me to get to step 2 and pick a file. After I do that I go back to the console and see error messages being printed again and again. While the error message beign constantly printed is identical, it also now adds another INFO before it, which is `Waiting for a DFU capable device` (screenshot below).

[image]

5. After I press reset button, messages keep keep being printed, but the keyboard stops working - the LED lights are on, but pressing buttons produces nothing (which Is why I needed a second keyboard).

6. Running `lsusb` gives me the following output (that's after resetting):

```
~ $ lsusb
Bus 008 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 007 Device 010: ID 09da:f613 A4Tech Co., Ltd. USB Device
Bus 007 Device 025: ID 16c0:0478 Van Ooijen Technische Informatica Teensy Halfkay Bootloader
Bus 007 Device 021: ID 1c4f:0026 SiGma Micro Keyboard
Bus 007 Device 020: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 007 Device 023: ID 258a:0024 SINOWEALTH Speedlink Sovos Gaming Mouse
Bus 007 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 003: ID 1e71:2007 NZXT NZXT USB Device
Bus 005 Device 002: ID 0b05:18f3 ASUSTek Computer, Inc. AURA LED Controller
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 002: ID 2109:0817 VIA Labs, Inc. USB3.0 Hub             
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 011: ID 0b0e:245e GN Netcom Jabra Link 370
Bus 003 Device 004: ID 0909:004e Audio-Technica Corp. USB2.0 Hub             
Bus 003 Device 003: ID 2109:2817 VIA Labs, Inc. USB2.0 Hub             
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 09da:f613 A4Tech Co., Ltd. 
Bus 001 Device 009: ID 258a:0024  
Bus 001 Device 007: ID 1c4f:0026 SiGma Micro Keyboard
Bus 001 Device 006: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
7. Reconnecting Ergoxdox makes it work again keeping my previous custom firmware. I ran lsusb once again after reconnecting Ergodox:

```
~ $ lsusb
Bus 008 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 007 Device 010: ID 09da:f613 A4Tech Co., Ltd. USB Device
Bus 007 Device 026: ID feed:1307 ZSA Ergodox EZ
Bus 007 Device 021: ID 1c4f:0026 SiGma Micro Keyboard
Bus 007 Device 020: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 007 Device 023: ID 258a:0024 SINOWEALTH Speedlink Sovos Gaming Mouse
Bus 007 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 003: ID 1e71:2007 NZXT NZXT USB Device
Bus 005 Device 002: ID 0b05:18f3 ASUSTek Computer, Inc. AURA LED Controller
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 002: ID 2109:0817 VIA Labs, Inc. USB3.0 Hub             
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 011: ID 0b0e:245e GN Netcom Jabra Link 370
Bus 003 Device 004: ID 0909:004e Audio-Technica Corp. USB2.0 Hub             
Bus 003 Device 003: ID 2109:2817 VIA Labs, Inc. USB2.0 Hub             
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 09da:f613 A4Tech Co., Ltd. 
Bus 001 Device 009: ID 258a:0024  
Bus 001 Device 007: ID 1c4f:0026 SiGma Micro Keyboard
Bus 001 Device 006: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

As you can see, in the first output Ergodox is absent, however I'm not sure why each output is still exactly 24 lines. Hope this helps.

kernel-io, 2020-11-27:
I'm experiencing the same problem while trying to flash my ErgoDox within Fedora 33

flurdy, 2020-12-31:
I was getting the same errors when trying to flash my new Moonlander running on Linux (PopOS 20.04), but I had not followed each step in https://github.com/zsa/wally/wiki/Linux-install. 

When I installed `sudo apt install libusb-dev` and extended as suggested the `/etc/udev/rules.d/50-wally.rules` it worked fine. Note the udev rules are more than the steps mentioned to enable Oryx training (https://github.com/zsa/wally/wiki/Live-training-on-Linux)