zsa/wally #21: Cannot flash on windows Wally 1.1.4 for ergodox

jpatrickpark, 2019-08-11:
It hangs at "Flashing firmware" page for longer than 14 minutes, cannot install new firmware.

Specs
- Windows 10
- Wally v1.1.4

Error message:
15:59:57 info Application UI / State initialized
15:59:58 info Probing compatible usb devices
15:59:58 info Found 1 compatible device(s)
16:00:06 info Starting Teensy Flash
16:00:06 info Waiting for a DFU capable device
...
16:00:17 info Waiting for a DFU capable device
16:00:17 error Error while sending firmware bytes: libusb: not found [code -5]

jpatrickpark, 2019-08-11:
Although this issue does not happen again when I launch the program for the second time, it succeeds to install. I will close this issue for now.

closed 2019-08-11 by jpatrickpark

acook, 2019-10-07:
This just happened to me too, fresh Wally install, same error with no more info. Even upon relaunching the program. I also tried to manually drop libusb.dll into the program path. It left my ErgoDoxEZ in an unusable state. I had to use Teensy Loader to put a new firmware on it.