zsa/wally #22: Graphical interface does not work on windows 7

Paillard, 2019-08-16:
Installed version 1.1.4, at the end, driver installer keep on searching compatible driver on windows update, and can't find one.

Tried to start wally as the installation ends well, but it stay on white prompt for the application.

Tried to install manually drivers, windows does not recognise them in the directory (wally/usb_driver)

Tried to use directly the wally-cli, but can't tell if it works as nothing tells you where is the reset button on the keyboard. Nor on the wiki, nor on the ergodox site.

Hopefully, the mac version seems to work but i didn't understant why a layout (colemak) I flashed is interpreted as azerty/qwerty (q/a inverted as I'm french. Maybe somethings to do with the windows keyboard layout ? Can't tell has there is no much information about it..

And by the way, the code needs comments to become more understandable so people can contribute.

fdidron, 2019-08-17:
Hi wally is not compatible with Windows 7 for the moment. Support is on our roadmap though.

Which board are you using ? If it's an Ergodox the reset button is on the right half, next to the three status leds. If it's a Planck it's located at the back of the keyboard.

Also if you are using an Ergodox you could try to use teensy loader on Windows 7 to flash https://www.pjrc.com/teensy/loader_win10.html

For the issue with your layout, can you shoot an email to contact@ergodox-ez.com ?

I'm keeping this open to track the win7 release

Paillard, 2019-08-17:
> Hi wally is not compatible with Windows 7 for the moment. Support is on our roadmap though.
Ok, I think it could be great to add some information about that in the readme description. something as :
`Supported os versions : 
- Windows 10 
- linux XXXX
- MacOs XXX

Roadmap : 
- support for windows 7
- ...`

I am on the ergodox, using Wally on mac give me the key about the reset button. I did retry using only the wally-cli on windows 7 knowing that, but I ran through multiple errors (globally, reset was detected, but firmware was not applied).

I'll try using teensy loader, could be usefull, thanks you for sharing about it.

Issue with layout seems to be inerant of the keyboard layout versus driver interpretation, story of scan codes and so. Forcing letters to be "French" letters, seems to be the workaround to avoid the driver replacing 'a' by 'q' and so on, I keep in mind the mail adress, just in case :)

Thanks for your quick response by the way, let me know if I can help you somehow.

fdidron, 2020-11-25:
@Paillard 

There's a new version of Wally for Windows here https://github.com/zsa/wally-win . Rewritten using native technologies, it should be compatible with Windows 7. Feel free to open a new issue in the new repo if needed.

closed 2020-11-25 by fdidron