zsa/wally #23: macOs 1.1.1 need to unplug keyboard after reflashing

newbould5, 2019-08-20:
Every time I reflash my ergodox-ez I manually need to unplug the usb and plug it back in or nothing will work. Everything is fine afterwards.

wally version 1.1.1
macOs 10.14.5

kescobo, 2019-12-05:
I'm having the same issue (same MacOS version). If I click the little terminal icon, I see

```
error
Error while rebooting device: libusb: i/o error [code -1]
```

I tried `brew install libusb`, but that doesn't seem to solve the issue.

biomaks, 2020-03-05:
I have exactly the same issue.
macOs: 10.15.3 (19D76)
wally: v1.1.3

kescobo, 2020-03-05:
Hmm - it was solved for me with Wally 1.1.3, but I'm still on OSX10.14.

biomaks, 2020-03-17:
yes I checked that, everything was updated

gliush, 2020-05-11:
The same problem for macos:10.15.4 and wally:1.1.3 
I have a switcher usb-c to usb-2 (default one from Apple), that might have caused the problem.

stantonk, 2020-06-03:
I'm also having the same issue, I can't successfully flash my keyboard with any new configurations:

macOS Mojave: 10.14.6
Wally: V1.1.3

Any ideas?

stantonk, 2020-06-03:
I got it to work if I plugged it directly into my macbook pro with a USB-A -> USB-C adapter. But when I had it plugged into a hub it failed. /shrug

DiscoRiver, 2020-06-03:
Apologies for the delay, I wanted to look into this sooner but had some corporate commitments to deal with. This is on my radar now so I'll see if I can get to the bottom of it.

DiscoRiver, 2020-06-03:
@stantonk @gliush @biomaks @kescobo 

If you're still experiencing this issue, could you please try again and send me the all the log output from a run where it fails? (click the little terminal icon at the bottom of the UI to see the log window)

It would seem this is caused by the ergoboot strategy, but I don't think that should ever generate the `Error while rebooting device: libusb: i/o error [code -1]` message, so I need more information.

Thanks!

DiscoRiver, 2020-06-05:
I just realised this is likely a duplicated for #3 and work should continue over there.

gliush, 2020-06-06:
hm, I tried several times and it always worked fine. I didn't manage it to work ever before (and I run it more than 100 times).