zsa/wally #24: panic while flashing ergodox

abscondment, 2019-08-22:
I'm using wally v1.1.1 (Linux) to flash my new ergodox ez. I get the same panic whichever path I go through.

Both the ui and the cli versions ask me to press the reset button, and once I do this the application crashes. They keyboard is then stuck in a non-functioning state until I re-plug it in. The layout remains unchanged after it starts up again.

Here's my cli panic; it's the same with the UI (happens under 1.1.0, too).

```
bribera@flask:~ 👻 $ wally-cli ~/Desktop/ergodox_ez_dvorak_e9XE.hex 
⠏ Press the reset button of your keyboard.panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x38 pc=0x4d4bb2]

goroutine 20 [running]:
github.com/google/gousb.(*Config).Close(0x0, 0x0, 0x0)
	/home/florian/dev/go/src/github.com/google/gousb/config.go:79 +0x42
github.com/zsa/wally/wally.TeensyFlash(0x7ffc15e1d442, 0x30, 0xc0000ca120)
	/home/florian/dev/go/src/github.com/zsa/wally/wally/teensy.go:80 +0xa08
created by main.main
	/home/florian/dev/go/src/github.com/zsa/wally/cli/main.go:39 +0x4ed
```

Am I missing something?

fdidron, 2019-08-23:
Hi can you tell me which distro and libusb version you are using ?

abscondment, 2019-08-23:
Sure, @fdidron:

```
bribera@flask:~ 👻 $ aptitude search libusb | grep '^i'
i A libusb-0.1-4 - userspace USB programming library
i  libusb-1.0-0 - userspace USB programming library
i  libusb-1.0-0-dev - userspace USB programming library development files
i A libusb-1.0-doc - documentation for userspace USB programming
i  libusb-dev - userspace USB programming library development files
i A libusbguard0 - USB device authorization policy framework - shared library
i A libusbmuxd4 - USB multiplexor daemon for iPhone and iPod Touch devices - library
i A libusbredirparser1 - Parser for the usbredir protocol (runtime)
bribera@flask:~ 👻 $ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.3 LTS
Release:	18.04
Codename:	bionic
```

kabo, 2020-03-28:
I've got this exact problem :(

```
$ sudo apt search libusb | grep installed
libftdi1-2/testing,unstable,now 1.4-2+b1 amd64 [installed,automatic]
libgusb2/testing,unstable,now 0.3.4-0.2 amd64 [installed,automatic]
libhidapi-libusb0/testing,unstable,now 0.9.0+dfsg-1 amd64 [installed,automatic]
libhpmud0/testing,unstable,now 3.20.3+dfsg0-1 amd64 [installed,automatic]
libusb-0.1-4/testing,unstable,stable,now 2:0.1.12-32 amd64 [installed,automatic]
libusb-1.0-0/testing,unstable,now 2:1.0.23-2 amd64 [installed,automatic]
libusb-1.0-0-dev/testing,unstable,now 2:1.0.23-2 amd64 [installed]
libusb-1.0-doc/testing,unstable,now 2:1.0.23-2 all [installed,automatic]
libusb-dev/testing,unstable,stable,now 2:0.1.12-32 amd64 [installed]
libusbguard0/testing,unstable,now 0.7.6+ds-1+b1 amd64 [installed,automatic]
libusbmuxd6/testing,unstable,now 2.0.1-2 amd64 [installed,automatic]
libusbredirhost1/testing,unstable,now 0.8.0-1+b1 amd64 [installed,automatic]
libusbredirparser1/testing,unstable,now 0.8.0-1+b1 amd64 [installed,automatic]

$ lsb_release -a
No LSB modules are available.
Distributor ID: Debian
Description:    Debian GNU/Linux bullseye/sid
Release:        testing
Codename:       bullseye
```

Getting a bunch of different errors with both 1.1.0 and 1.1.1

```
kabo@esk:~/Downloads$ ./wally2
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x38 pc=0x542802]

goroutine 19 [running]:
github.com/google/gousb.(*Config).Close(0x0, 0x0, 0x0)
        /home/florian/dev/go/src/github.com/google/gousb/config.go:79 +0x42
github.com/zsa/wally/wally.TeensyFlash(0xc00040e2c0, 0x3d, 0xc000104000)
        /home/florian/dev/go/src/github.com/zsa/wally/wally/teensy.go:80 +0xa08
created by github.com/zsa/wally/wally.(*State).FlashFirmware
        /home/florian/dev/go/src/github.com/zsa/wally/wally/state.go:163 +0xa5
kabo@esk:~/Downloads$ ./wally2
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x38 pc=0x542802]

goroutine 6 [running]:
github.com/google/gousb.(*Config).Close(0x0, 0x0, 0x0)
        /home/florian/dev/go/src/github.com/google/gousb/config.go:79 +0x42
github.com/zsa/wally/wally.TeensyFlash(0xc000020800, 0x3d, 0xc000100000)
        /home/florian/dev/go/src/github.com/zsa/wally/wally/teensy.go:80 +0xa08
created by github.com/zsa/wally/wally.(*State).FlashFirmware
        /home/florian/dev/go/src/github.com/zsa/wally/wally/state.go:163 +0xa5
kabo@esk:~/Downloads$ ./wally2
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x38 pc=0x542802]

goroutine 19 [running]:
github.com/google/gousb.(*Config).Close(0x0, 0x0, 0x0)
        /home/florian/dev/go/src/github.com/google/gousb/config.go:79 +0x42
github.com/zsa/wally/wally.TeensyFlash(0xc00040c2c0, 0x3d, 0xc000102000)
        /home/florian/dev/go/src/github.com/zsa/wally/wally/teensy.go:80 +0xa08
created by github.com/zsa/wally/wally.(*State).FlashFirmware
        /home/florian/dev/go/src/github.com/zsa/wally/wally/state.go:163 +0xa5
kabo@esk:~/Downloads$ ./wally2
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x38 pc=0x542802]

goroutine 34 [running]:
github.com/google/gousb.(*Config).Close(0x0, 0x0, 0x0)
        /home/florian/dev/go/src/github.com/google/gousb/config.go:79 +0x42
github.com/zsa/wally/wally.TeensyFlash(0xc000416080, 0x3d, 0xc0000fa000)
[... 71 lines cut ...]
created by github.com/zsa/wally/wally.(*State).FlashFirmware
        /home/florian/dev/go/src/github.com/zsa/wally/wally/state.go:163 +0xa5
kabo@esk:~/Downloads$ ./wally2 ergodox_ez_first-colemak_K0db7_eW4mE.hex
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x38 pc=0x542802]

goroutine 4 [running]:
github.com/google/gousb.(*Config).Close(0x0, 0x0, 0x0)
        /home/florian/dev/go/src/github.com/google/gousb/config.go:79 +0x42
github.com/zsa/wally/wally.TeensyFlash(0x7ffca9216c10, 0x28, 0xc00010a000)
        /home/florian/dev/go/src/github.com/zsa/wally/wally/teensy.go:80 +0xa08
created by github.com/zsa/wally/wally.(*State).FlashFirmware
        /home/florian/dev/go/src/github.com/zsa/wally/wally/state.go:163 +0xa5
kabo@esk:~/Downloads$ ./wally2 ergodox_ez_first-colemak_K0db7_eW4mE.hex
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x38 pc=0x542802]

goroutine 18 [running]:
github.com/google/gousb.(*Config).Close(0x0, 0x0, 0x0)
        /home/florian/dev/go/src/github.com/google/gousb/config.go:79 +0x42
github.com/zsa/wally/wally.TeensyFlash(0x7fffd186dc10, 0x28, 0xc000104000)
        /home/florian/dev/go/src/github.com/zsa/wally/wally/teensy.go:80 +0xa08
created by github.com/zsa/wally/wally.(*State).FlashFirmware
        /home/florian/dev/go/src/github.com/zsa/wally/wally/state.go:163 +0xa5
```

kabo, 2020-03-28:
In my case there was something funky going on with `usbguard`. My guess is that the keyboard tries to present it self as another device during the flash process.

Stopping usbguard didn't work either ([bug](https://github.com/USBGuard/usbguard/issues/367)).

Disabling usbguard, rebooting, flashing, enabling usbguard, rebooting, fixing the usbguard rules because the keyboard name is now `Ergodox EZ` instead of `ErgoDox EZ`, rebooting again, is what worked for me :)