zsa/wally #25: Keyboard not detected when connected via USB Dock

Deltatiger, 2019-08-28:
I am using a Dell USB C dock for my laptop and the EZ is connected to the dock. When I try wally, the keyboard is not detected until it is connected directly to the Laptop.

Is this the expected behavior?

fdidron, 2019-09-12:
@Deltatiger There have been some issues with some USB dongles / hubs. Can you share the OS you are using and also which keyboard you are using (Planck or Ergodox) ? Thanks

sajjad, 2020-05-02:
I have exactly the same issue with my ergodox.
Dell XPS 15 9560 running Win 10 Version 1909 Build 18363.815
Using the USB ports on a Razer Core X Chroma

mykter, 2020-07-22:
I had intermittent issues with my Ergodox and using a hub, on Linux and Windows. Sometimes I even experienced the right half of the keyboard working and the left seemingly being without power (no lights). I've resorted to a direct connection (which is more effort than it sounds, as I have to switch it between machines several times a day).

fdidron, 2020-11-25:
Hi @mykter  @Deltatiger  @sajjad 

There's a new version of Wally for Windows here https://github.com/zsa/wally-win . Rewritten using native technologies, it should fix that kind of issues. Feel free to open a new issue in the new repo if needed.

closed 2020-11-25 by fdidron