zsa/wally #26: Doesn't make any progress flashing keyboard

thanatos, 2019-09-03:
It gets to "Flashing firmware" and says "Please don't unplug your board" (which I'm going to have to do here after I finish this bug report 😫)

The built-in "terminal" says,
```
error  Error while sending firmware bytes: libusb: pipe error [code -9]
```

This is on OS X, as downloaded from Ergo-Dox ([here](https://ergodox-ez.com/pages/wally))

clayzermk1, 2019-09-11:
Wally requires libusb on macOS per the readme, but it looks like it doesn't do a check for it in the binary on macOS since it lets me begin flashing without having it. I'm in the same situation, brand new EZ, can't flash by just installing Wally. Overall this creates a bad user experience.

clayzermk1, 2019-09-11:
As a workaround, one can: close Wally, `brew install libusb`, start Wally, reattempt the flash, unplug and replug the keyboard to simulate a reboot of the keyboard. The pipe error above still shows, but at least it flashes.

Version info:
macOS 10.14.6 (18G87)
libusb 1.0.22
Wally 1.1.0 (downloaded from Ergodox website)

fdidron, 2019-09-12:
@clayzermk1 Do you happen to have the keyboard plugged via a usb dongle or a hub ? I am considering removing the keyboard phase from Wally and make it so it asks users to unplug / plug the keyboard after flashing since this issue seems to only affect some (not all) dongles / hubs

fdidron, 2019-09-12:
@thanatos can you share your OS version as well as the keyboard model you are using ? (Planck or Ergodox ? )

abhidp, 2019-11-24:
Although the flashing tool gets stuck at `Flashing Firmware, Please do not unplug your board.` and the terminal log says `error
Error while rebooting device: libusb: i/o error [code -1]` the hex has already been applied to your board even without rebooting. Just ignore the error messages. You can start typing with the new layout straight away.

kescobo, 2020-01-29:
I was having this problem as well, but seems to be fixed with the recent update 👍

simplycycling, 2020-03-28:
I'm seeing the same issue with the version I just downloaded today, Ergodox-EX, current MacOS verion.

phette, 2020-04-01:
It seemed to work for me despite the error message.  Not sure if anything is actually broken or incomplete.

goalaleo, 2020-07-16:
I'm also on MacOS and the first time I tried to flash my keyboard, it didn't work at all - loading bar was stuck at 0%, and console showed `error while sending firmware bytes: libusb: pipe error`.

I replugged the keyboard, and now the flashing worked, but there's an error at the end when it tries to reboot
[image]

The UI is stuck at 100%, but it seems like the layout was applied
[image]

dc5048, 2020-08-07:
Same issue on a 2019 MB running Catalina 10.15.6 + USB hub.
I get the -1 i/o error.
Flash does appear to have been successful though.

haukepauke, 2020-12-01:
I have a similar issue.
Tried to flash my keyboard. Got the error "Error while sending firmware bytes: libusb: i/o error [code -1]". 
Since then the keyboard stopped working. Starting wally with connected keyboard does not work anymore. 

Starting Wally without keyboard connected works. When connecting the keyboard after wally has started it is recognized. However when selecting a keyboard layout, wally starts to flash right away without waiting for pushing the reset pin. Flashing the keyboard  fails with the error message above.

I'm on Ubuntu 20.04., now using my old keyboard again. Would appreciate any help.

thomasluquet, 2021-01-10:
Same on linux...