zsa/wally #29: macOS crash (fails to open)

xavdid, 2019-09-19:
I just downloaded this app (via `brew cask install zsa-wally`, if that matters) and I've been unable to open it. My keyboard isn't attached (I wanted to open the app and see what happened).

Here's the crash log:

```
Process:               Wally [59632]
Path:                  /Applications/Wally.app/Contents/MacOS/Wally
Identifier:            com.zsa.wally
Version:               0
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           Wally [59632]
User ID:               501

Date/Time:             2019-09-19 12:03:25.522 -0500
OS Version:            Mac OS X 10.14.5 (18F132)
Report Version:        12
Bridge OS Version:     3.0 (14Y901)
Anonymous UUID:        4BA15E8D-CED9-8FFC-1CE8-2B9B647D8182

Sleep/Wake UUID:       E1E69E25-2858-4F1E-9521-44847B84A324

Time Awake Since Boot: 340000 seconds
Time Since Wake:       6100 seconds

System Integrity Protection: enabled

Crashed Thread:        0

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    DYLD, [0x1] Library missing

Application Specific Information:
dyld: launch, loading dependent libraries

Dyld Error Message:
  Library not loaded: /usr/local/opt/libusb/lib/libusb-1.0.0.dylib
  Referenced from: /Applications/Wally.app/Contents/MacOS/Wally
  Reason: image not found

Binary Images:
         0x4000000 -          0x4328fe7 +com.zsa.wally (0) <DE4622D3-E9A1-363A-A068-2A275CF64581> /Applications/Wally.app/Contents/MacOS/Wally
         0xda5d000 -          0xdac76ef  dyld (655.1.1) <CE635DB2-D47E-3C05-A0A3-6BD982E7E750> /usr/lib/dyld

```

As it notes, I'm on `Mac OS X 10.14.5 (18F132)` (Mojave) if that's helpful. 

Running `brew install libusb` fixed the problem. Is there a way that can be included (or declared as a dependency)?

fdidron, 2019-09-23:
@xavdid I will update Wally to include the depency this week

fdidron, 2019-09-25:
@xavdid I released version 1.1.3 of Wally that embeds libusb with the Gui App, there is a PR open to update Wally to v1.1.3 on brew cask https://github.com/Homebrew/homebrew-cask-drivers/pull/1210

closed 2019-09-25 by fdidron

xavdid, 2019-09-25:
perfect, thank you!