zsa/wally #31: open --version: no such file or directory

ngirard, 2019-09-25:
Just downloaded & run latest Wally V1.1.1 on Ubuntu 19.04.

Got the following messages:
```
17:54:09 info   Found 1 compatible device(s)
17:54:09 info   Probing compatible usb devices
17:54:06 info   Application UI / State initialized
17:54:06 error  Error while opening firmware: open --version: no such file or directory
```

DiscoRiver, 2020-02-26:
The name of the file the application is trying to open is "--version", so I imagine there was an issue with the command. 

Suggest closing.

ngirard, 2020-02-26:
I'd rather suggest *implementing* such `--version` option.

Perhaps my title was misleading.

DiscoRiver, 2020-02-26:
Ah, gotcha. I'll get an update in.

Thanks!

closed 2020-06-15 by fdidron