zsa/wally #34: Wally is from an unidentified developer

JasonMFry, 2019-10-29:
I'm getting the below error when trying to install wally from the official website, brew, or when downloading the latest release from github. I'm on my work machine and am not supposed to install apps from unidentified developers. I thought to compile the source but I'm having separate issues with that.

[image]

fdidron, 2019-11-05:
Hi @JasonMFry I'm currently working on Wally rewrite using Swift / XCode that will be signed for submission on the Mac Store, it is going to take a while though. Which version of OS X are you using ? Is it for an Ergodox or a Planck ?

JasonMFry, 2019-11-06:
@fdidron I'm on 10.14.6 using an ergodox shine. Thanks a lot! I can always compile from source, as I mentioned I was having issues with that which I can't remember right now. I think this is an important thing, but most people should be satisfied by compiling from source I would think.

nmelin, 2019-12-18:
Is there a timeline for this yet? Also,  is there a documented workaround for flashing from macOS Catalina?  (see also, #37)

@fdidron

fdidron, 2020-01-10:
I updated the build scripts and README for OSX, can you pull the latest commit from master then follow in order:

https://github.com/zsa/wally#pre-requisites-for-all-os
https://github.com/zsa/wally#mac-os

toddhodgson, 2020-01-11:
Is there any timing on this?

fdidron, 2020-06-16:
Hi everyone, sorry for the lack of activity in here, things have been on a crazy side lately.
Regarding this issue, as well as being able to compile locally #37 #27  here are the news:
1. We are in the process of getting Wally signed by Apple, hopefully it should be done around next week.
2. I am in the process of porting Wally to https://wails.app/ . It uses the same technology than our current code, but makes the development / build process much simpler and it is actively maintained. For this to happen, I need to implement this feature first: https://github.com/wailsapp/wails/issues/436

fdidron, 2020-06-23:
Hello, the binary was signed https://github.com/zsa/wally/releases/download/1.1.3-osx/wally-osx-1.1.3.dmg . Can you download it and give it a try ?

JasonMFry, 2020-06-25:
> Hello, the binary was signed https://github.com/zsa/wally/releases/download/1.1.3-osx/wally-osx-1.1.3.dmg . Can you download it and give it a try ?

I actually was able to download it several days ago, I don't remember when exactly but I think it was at least 10 days ago if not longer. But Apple didn't complain that it was unsigned when I tried to open it. But maybe I'm misremembering and it was less than 10 days ago. Either way it's working, thanks!

closed 2020-06-25 by JasonMFry