zsa/wally #37: Errors trying to compile release build on macos

JasonMFry, 2019-11-06:
## Why

I want to compile a release build because of [this issue](https://github.com/zsa/wally/issues/34). 

## How to Reproduce

1. Buy a Macbook and install macOS 10.14.6
1. cd to `$GOPATH/src/github.com`
1. clone this repo
1. run `brew install libusb`
1. run `sh build.osx.sh`

## Error Output

```sh
yarn run v1.19.1
$ cross-env NODE_ENV=production webpack
/bin/sh: cross-env: command not found
error Command failed with exit code 127.
info Visit https://yarnpkg.com/en/docs/cli/run for documentation about this command.
build.osx.sh: line 5: go-bindata: command not found
ui.go:4:2: cannot find package "github.com/fdidron/webview" in any of:
	/usr/local/go/src/github.com/fdidron/webview (from $GOROOT)
	/Users/j/go/src/github.com/fdidron/webview (from $GOPATH)
ui.go:5:2: cannot find package "github.com/zsa/wally/wally" in any of:
	/usr/local/go/src/github.com/zsa/wally/wally (from $GOROOT)
	/Users/j/go/src/github.com/zsa/wally/wally (from $GOPATH)
cli/main.go:5:2: cannot find package "github.com/caarlos0/spin" in any of:
	/usr/local/go/src/github.com/caarlos0/spin (from $GOROOT)
	/Users/j/go/src/github.com/caarlos0/spin (from $GOPATH)
cli/main.go:6:2: cannot find package "github.com/logrusorgru/aurora" in any of:
	/usr/local/go/src/github.com/logrusorgru/aurora (from $GOROOT)
	/Users/j/go/src/github.com/logrusorgru/aurora (from $GOPATH)
cli/main.go:11:2: cannot find package "github.com/zsa/wally/wally" in any of:
	/usr/local/go/src/github.com/zsa/wally/wally (from $GOROOT)
	/Users/j/go/src/github.com/zsa/wally/wally (from $GOPATH)
cli/main.go:7:2: cannot find package "gopkg.in/cheggaaa/pb.v1" in any of:
	/usr/local/go/src/gopkg.in/cheggaaa/pb.v1 (from $GOROOT)
	/Users/j/go/src/gopkg.in/cheggaaa/pb.v1 (from $GOPATH)
build.osx.sh: line 12: upx: command not found
```

## Troubleshooting

After running `npm install cross-env -g` and running `sh build.osx.sh` I get this output:

```sh
yarn run v1.19.1
$ cross-env NODE_ENV=production webpack
events.js:187
      throw er; // Unhandled 'error' event
      ^

Error: spawn webpack ENOENT
    at Process.ChildProcess._handle.onexit (internal/child_process.js:264:19)
    at onErrorNT (internal/child_process.js:456:16)
    at processTicksAndRejections (internal/process/task_queues.js:80:21)
Emitted 'error' event on ChildProcess instance at:
    at Process.ChildProcess._handle.onexit (internal/child_process.js:270:12)
    at onErrorNT (internal/child_process.js:456:16)
    at processTicksAndRejections (internal/process/task_queues.js:80:21) {
  errno: 'ENOENT',
  code: 'ENOENT',
  syscall: 'spawn webpack',
  path: 'webpack',
  spawnargs: []
}
error Command failed with exit code 1.
info Visit https://yarnpkg.com/en/docs/cli/run for documentation about this command.
build.osx.sh: line 5: go-bindata: command not found
ui.go:4:2: cannot find package "github.com/fdidron/webview" in any of:
	/usr/local/go/src/github.com/fdidron/webview (from $GOROOT)
	/Users/j/go/src/github.com/fdidron/webview (from $GOPATH)
ui.go:5:2: cannot find package "github.com/zsa/wally/wally" in any of:
	/usr/local/go/src/github.com/zsa/wally/wally (from $GOROOT)
	/Users/j/go/src/github.com/zsa/wally/wally (from $GOPATH)
cli/main.go:5:2: cannot find package "github.com/caarlos0/spin" in any of:
	/usr/local/go/src/github.com/caarlos0/spin (from $GOROOT)
	/Users/j/go/src/github.com/caarlos0/spin (from $GOPATH)
cli/main.go:6:2: cannot find package "github.com/logrusorgru/aurora" in any of:
	/usr/local/go/src/github.com/logrusorgru/aurora (from $GOROOT)
	/Users/j/go/src/github.com/logrusorgru/aurora (from $GOPATH)
cli/main.go:11:2: cannot find package "github.com/zsa/wally/wally" in any of:
	/usr/local/go/src/github.com/zsa/wally/wally (from $GOROOT)
	/Users/j/go/src/github.com/zsa/wally/wally (from $GOPATH)
cli/main.go:7:2: cannot find package "gopkg.in/cheggaaa/pb.v1" in any of:
	/usr/local/go/src/gopkg.in/cheggaaa/pb.v1 (from $GOROOT)
	/Users/j/go/src/gopkg.in/cheggaaa/pb.v1 (from $GOPATH)
build.osx.sh: line 12: upx: command not found
```

Something is wrong, I'm sure you don't intend for me to install these packages one-by-one.

chris-kahn, 2020-01-08:
Is there no one monitoring this repo? Any update?

fdidron, 2020-01-10:
Hi, sorry for the late replies on this,

I updated the build scripts and README for OSX, can you pull the latest commit from master then follow in order:

https://github.com/zsa/wally#pre-requisites-for-all-os
https://github.com/zsa/wally#mac-os

JasonMFry, 2020-01-10:
@fdidron thanks for working on this. I'm still getting errors :(

1. clone this repo fresh
1. install dep with `brew install dep`
1. install go-bindata with `go get -u github.com/jteeuwen/go-bindata/...`
1. install libusb with `brew install libusb`
1. Install upx with `brew: brew install upx`
1. run `sh build.osx.sh`

```
yarn run v1.19.1
$ cross-env NODE_ENV=production webpack
/bin/sh: cross-env: command not found
error Command failed with exit code 127.
info Visit https://yarnpkg.com/en/docs/cli/run for documentation about this command.
bindata: Failed to stat input path 'ui/build/index.dist.js': lstat ui/build/index.dist.js: no such file or directory
wally/dfu.go:6:2: cannot find package "github.com/google/gousb" in any of:
	/usr/local/Cellar/go/1.13.6/libexec/src/github.com/google/gousb (from $GOROOT)
	/Users/j/go/src/github.com/google/gousb (from $GOPATH)
wally/teensy.go:6:2: cannot find package "github.com/marcinbor85/gohex" in any of:
	/usr/local/Cellar/go/1.13.6/libexec/src/github.com/marcinbor85/gohex (from $GOROOT)
	/Users/j/go/src/github.com/marcinbor85/gohex (from $GOPATH)
                       Ultimate Packer for eXecutables
                          Copyright (C) 1996 - 2018
UPX 3.95        Markus Oberhumer, Laszlo Molnar & John Reiser   Aug 26th 2018

        File size         Ratio      Format      Name
   --------------------   ------   -----------   -----------
upx: dist/osx/Wally.app/Contents/MacOS/Wally: FileNotFoundException: dist/osx/Wally.app/Contents/MacOS/Wally: No such file or directory

Packed 0 files.
cli/main.go:5:2: cannot find package "github.com/caarlos0/spin" in any of:
	/usr/local/Cellar/go/1.13.6/libexec/src/github.com/caarlos0/spin (from $GOROOT)
	/Users/j/go/src/github.com/caarlos0/spin (from $GOPATH)
wally/dfu.go:6:2: cannot find package "github.com/google/gousb" in any of:
	/usr/local/Cellar/go/1.13.6/libexec/src/github.com/google/gousb (from $GOROOT)
	/Users/j/go/src/github.com/google/gousb (from $GOPATH)
cli/main.go:6:2: cannot find package "github.com/logrusorgru/aurora" in any of:
	/usr/local/Cellar/go/1.13.6/libexec/src/github.com/logrusorgru/aurora (from $GOROOT)
	/Users/j/go/src/github.com/logrusorgru/aurora (from $GOPATH)
wally/teensy.go:6:2: cannot find package "github.com/marcinbor85/gohex" in any of:
	/usr/local/Cellar/go/1.13.6/libexec/src/github.com/marcinbor85/gohex (from $GOROOT)
	/Users/j/go/src/github.com/marcinbor85/gohex (from $GOPATH)
cli/main.go:7:2: cannot find package "gopkg.in/cheggaaa/pb.v1" in any of:
	/usr/local/Cellar/go/1.13.6/libexec/src/gopkg.in/cheggaaa/pb.v1 (from $GOROOT)
	/Users/j/go/src/gopkg.in/cheggaaa/pb.v1 (from $GOPATH)
                       Ultimate Packer for eXecutables
                          Copyright (C) 1996 - 2018
UPX 3.95        Markus Oberhumer, Laszlo Molnar & John Reiser   Aug 26th 2018

        File size         Ratio      Format      Name
   --------------------   ------   -----------   -----------
upx: dist/osx/wally-cli: FileNotFoundException: dist/osx/wally-cli: No such file or directory

Packed 0 files.
```

1. `npm install cross-env -g`
1. `sh build.osx.sh`

```
yarn run v1.19.1
$ cross-env NODE_ENV=production webpack
events.js:282
      throw er; // Unhandled 'error' event
      ^

Error: spawn webpack ENOENT
    at Process.ChildProcess._handle.onexit (internal/child_process.js:267:19)
    at onErrorNT (internal/child_process.js:469:16)
[... 31 lines cut ...]
	/Users/j/go/src/github.com/caarlos0/spin (from $GOPATH)
wally/dfu.go:6:2: cannot find package "github.com/google/gousb" in any of:
	/usr/local/Cellar/go/1.13.6/libexec/src/github.com/google/gousb (from $GOROOT)
	/Users/j/go/src/github.com/google/gousb (from $GOPATH)
cli/main.go:6:2: cannot find package "github.com/logrusorgru/aurora" in any of:
	/usr/local/Cellar/go/1.13.6/libexec/src/github.com/logrusorgru/aurora (from $GOROOT)
	/Users/j/go/src/github.com/logrusorgru/aurora (from $GOPATH)
wally/teensy.go:6:2: cannot find package "github.com/marcinbor85/gohex" in any of:
	/usr/local/Cellar/go/1.13.6/libexec/src/github.com/marcinbor85/gohex (from $GOROOT)
	/Users/j/go/src/github.com/marcinbor85/gohex (from $GOPATH)
cli/main.go:7:2: cannot find package "gopkg.in/cheggaaa/pb.v1" in any of:
	/usr/local/Cellar/go/1.13.6/libexec/src/gopkg.in/cheggaaa/pb.v1 (from $GOROOT)
	/Users/j/go/src/gopkg.in/cheggaaa/pb.v1 (from $GOPATH)
                       Ultimate Packer for eXecutables
                          Copyright (C) 1996 - 2018
UPX 3.95        Markus Oberhumer, Laszlo Molnar & John Reiser   Aug 26th 2018

        File size         Ratio      Format      Name
   --------------------   ------   -----------   -----------
upx: dist/osx/wally-cli: FileNotFoundException: dist/osx/wally-cli: No such file or directory

Packed 0 files.
```

There is a file in `./dist/osx` so I tried `open dist/osx/Wally.app` and get `The application cannot be opened because its executable is missing.`

fdidron, 2020-01-11:
Hi Jason,

There seems to be two errors on the logs above, can you try to: 
1. npm i -g webpack
2. make sure you run dep ensure between step 2 and 3

I will update the docs accordingly, if it works

JasonMFry, 2020-01-25:
@fdidron That may have worked. I still got some errors (see below), but I did see this:

```
wally-cli  6/6  [****************************************
3789392 ->   1892368   49.94%   macho/amd64   wally-cli

Packed 1 file.
```

Wally.app and wally-cli are both in ./dist/osx, but I'm not sure how to use them. I've tried `open Wally.app` but I get `The application cannot be opened because its executable is missing.` When trying to use wally-cli, `which wally` and `which wally-cli` return `wally not found` or `wally-cli not found`. So maybe it was successfully built but I don't know how to properly run it?

## Errors
```
$ sh build.osx.sh
yarn run v1.19.1
$ cross-env NODE_ENV=production webpack
One CLI for webpack must be installed. These are recommended choices, delivered as separate packages:
 - webpack-cli (https://github.com/webpack/webpack-cli)
   The original webpack full-featured CLI.
We will use "yarn" to install the CLI via "yarn add -D".
Do you want to install 'webpack-cli' (yes/no): yes
Installing 'webpack-cli' (running 'yarn add -D webpack-cli')...
[1/4] 🔍  Resolving packages...
[2/4] 🚚  Fetching packages...
[3/4] 🔗  Linking dependencies...
[4/4] 🔨  Building fresh packages...
[-/3] ⠠ waiting...
[-/3] ⠠ waiting...
warning Error running install script for optional dependency: "/Users/j/go/src/github.com/wally/ui/node_modules/fsevents: Command failed.
Exit code: 1
Command: node install
Arguments:
Directory: /Users/j/go/src/github.com/wally/ui/node_modules/fsevents
Output:
node-pre-gyp info it worked if it ends with ok
node-pre-gyp info using node-pre-gyp@0.10.3
node-pre-gyp info using node@13.7.0 | darwin | x64
node-pre-gyp WARN Using needle for node-pre-gyp https download
node-pre-gyp info check checked for \"/Users/j/go/src/github.com/wally/ui/node_modules/fsevents/lib/binding/Release/node-v79-darwin-x64/fse.node\" (not found)
node-pre-gyp http GET https://fsevents-binaries.s3-us-west-2.amazonaws.com/v1.2.7/fse-v1.2.7-node-v79-darwin-x64.tar.gz
node-pre-gyp http 404 https://fsevents-binaries.s3-us-west-2.amazonaws.com/v1.2.7/fse-v1.2.7-node-v79-darwin-x64.tar.gz
node-pre-gyp WARN Tried to download(404): https://fsevents-binaries.s3-us-west-2.amazonaws.com/v1.2.7/fse-v1.2.7-node-v79-darwin-x64.tar.gz
node-pre-gyp WARN Pre-built binaries not found for fsevents@1.2.7 and node@13.7.0 (node-v79 ABI, unknown) (falling back to source compile with node-gyp)
node-pre-gyp http 404 status code downloading tarball https://fsevents-binaries.s3-us-west-2.amazonaws.com/v1.2.7/fse-v1.2.7-node-v79-darwin-x64.tar.gz
gyp info it worked if it ends with ok
gyp info using node-gyp@5.0.5
gyp info using node@13.7.0 | darwin | x64
gyp info ok
gyp info it worked if it ends with ok
gyp info using node-gyp@5.0.5
gyp info using node@13.7.0 | darwin | x64
gyp info find Python using Python version 3.7.6 found at \"/usr/local/opt/python/bin/python3.7\"
gyp info spawn /usr/local/opt/python/bin/python3.7
gyp info spawn args [
gyp info spawn args   '/usr/local/Cellar/node/13.7.0/libexec/lib/node_modules/npm/node_modules/node-gyp/gyp/gyp_main.py',
gyp info spawn args   'binding.gyp',
gyp info spawn args   '-f',
gyp info spawn args   'make',
gyp info spawn args   '-I',
gyp info spawn args   '/Users/j/go/src/github.com/wally/ui/node_modules/fsevents/build/config.gypi',
gyp info spawn args   '-I',
gyp info spawn args   '/usr/local/Cellar/node/13.7.0/libexec/lib/node_modules/npm/node_modules/node-gyp/addon.gypi',
gyp info spawn args   '-I',
gyp info spawn args   '/Users/j/Library/Caches/node-gyp/13.7.0/include/node/common.gypi',
gyp info spawn args   '-Dlibrary=shared_library',
gyp info spawn args   '-Dvisibility=default',
gyp info spawn args   '-Dnode_root_dir=/Users/j/Library/Caches/node-gyp/13.7.0',
gyp info spawn args   '-Dnode_gyp_dir=/usr/local/Cellar/node/13.7.0/libexec/lib/node_modules/npm/node_modules/node-gyp',
gyp info spawn args   '-Dnode_lib_file=/Users/j/Library/Caches/node-gyp/13.7.0/<(target_arch)/node.lib',
gyp info spawn args   '-Dmodule_root_dir=/Users/j/go/src/github.com/wally/ui/node_modules/fsevents',
[... 74 lines cut ...]
    at require (internal/modules/cjs/helpers.js:72:18)
    at /usr/local/lib/node_modules/webpack/bin/webpack.js:143:5
    at processTicksAndRejections (internal/process/task_queues.js:97:5) {
  code: 'MODULE_NOT_FOUND',
  requireStack: [ '/usr/local/lib/node_modules/webpack/bin/webpack.js' ]
}
error Command failed with exit code 1.
info Visit https://yarnpkg.com/en/docs/cli/run for documentation about this command.
bindata: Failed to stat input path 'ui/build/index.dist.js': lstat ui/build/index.dist.js: no such file or directory
# github.com/wally
./ui_dist.go:13:11: undefined: wally.NewState
./ui_dist.go:26:15: undefined: MustAsset
                       Ultimate Packer for eXecutables
                          Copyright (C) 1996 - 2020
UPX 3.96        Markus Oberhumer, Laszlo Molnar & John Reiser   Jan 23rd 2020

        File size         Ratio      Format      Name
   --------------------   ------   -----------   -----------
upx: dist/osx/Wally.app/Contents/MacOS/Wally: FileNotFoundException: dist/osx/Wally.app/Contents/MacOS/Wally: No such file or directory

Packed 0 files.
                       Ultimate Packer for eXecutables
                          Copyright (C) 1996 - 2020
UPX 3.96        Markus Oberhumer, Laszlo Molnar & John Reiser   Jan 23rd 2020
```