zsa/wally #38: Couldn't claim usb interface, is your board flashed with the latest firmware version?

javinievas, 2019-12-10:
Getting this error message after selecting the keyboard in the chrome popup:

[image]

I've flashed the keyboard with the latest firmware version (v15) 

I've followed the step-by-step tutorial and the linux instructions related to the udev rules.

Got:

- Ubuntu 16.04 
- Chrome 78
- Ergodox EZ Glow

I'm posting this issue here, since is where the associated wiki lives, but maybe it is not the right place (?)

fdidron, 2019-12-10:
Hi @javinievas Did you assign the [Oryx key](https://configure.ergodox-ez.com/planck-ez/layouts/dPvrN/YEdXw/3/11/keys) to your layout ?

javinievas, 2019-12-11:
That did the trick!

Sorry!

fdidron, 2019-12-11:
Great to hear @javinievas and no need to be sorry, the error message should include that info, I am updating it.

closed 2019-12-11 by fdidron

JessicaSachs, 2020-11-23:
Hmm, would be helpful to know what the oryx key was from the error message. I'm googling around.