zsa/wally #39: Unable to build CLI on macOS

JasonMFry, 2019-12-23:
I'm trying to build the cli version on macOS 10.14.6. Either I have something setup terribly wrong, or the instructions to build this are terribly wrong. FWIW `echo "$GOPATH"` => `/Users/j/go`. I cloned the repo into `$GOPATH/src/github.com`

## Steps to Reproduce

1. Clone the repo: `cd $GOPATH/src/github.com && git clone git@github.com:zsa/wally.git`
1. Build: `go build cli/main.go -o wally-cli`
1. Get error: `can't load package: named files must be .go files: -o`
1. Try the following commands out of desperation `go build cli/main.go -o wally-cli.go -i`, `go build cli/main.go`, `go build -i cli/main.go`. The last two commands result in this new error:
```
cli/main.go:5:2: cannot find package "github.com/caarlos0/spin" in any of:
	/usr/local/go/src/github.com/caarlos0/spin (from $GOROOT)
	/Users/j/go/src/github.com/caarlos0/spin (from $GOPATH)
wally/dfu.go:6:2: cannot find package "github.com/google/gousb" in any of:
	/usr/local/go/src/github.com/google/gousb (from $GOROOT)
	/Users/j/go/src/github.com/google/gousb (from $GOPATH)
cli/main.go:6:2: cannot find package "github.com/logrusorgru/aurora" in any of:
	/usr/local/go/src/github.com/logrusorgru/aurora (from $GOROOT)
	/Users/j/go/src/github.com/logrusorgru/aurora (from $GOPATH)
wally/teensy.go:6:2: cannot find package "github.com/marcinbor85/gohex" in any of:
	/usr/local/go/src/github.com/marcinbor85/gohex (from $GOROOT)
	/Users/j/go/src/github.com/marcinbor85/gohex (from $GOPATH)
cli/main.go:7:2: cannot find package "gopkg.in/cheggaaa/pb.v1" in any of:
	/usr/local/go/src/gopkg.in/cheggaaa/pb.v1 (from $GOROOT)
	/Users/j/go/src/gopkg.in/cheggaaa/pb.v1 (from $GOPATH)
```

What am I doing wrong here?

fdidron, 2020-01-10:
Hi @JasonMFry ,

I updated the build scripts and README for OSX, can you pull the latest commit from master then follow in order:

https://github.com/zsa/wally#pre-requisites-for-all-os
https://github.com/zsa/wally#mac-os

revmischa, 2020-05-12:
On master I get:
```
❯ go build cli/main.go -o wally-cli
named files must be .go files
❯ go build -o wally-ci cli/main.go

# command-line-arguments
cli/main.go:22:46: undefined: wally.GetAppVersion

❯ go build -tags dev -o wally

# github.com/wally/vendor/github.com/fdidron/webview
In file included from _cgo_export.c:4:
In file included from webview.go:27:
vendor/github.com/fdidron/webview/webview.h:84:3: warning: 'WebView' is deprecated: first deprecated in macOS 10.14 - No longer supported; please adopt WKWebView. [-Wdeprecated-declarations]
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/System/Library/Frameworks/WebKit.framework/Headers/WebView.h:119:12: note: 'WebView' has been explicitly marked deprecated here
In file included from _cgo_export.c:4:
In file included from webview.go:27:
vendor/github.com/fdidron/webview/webview.h:1691:17: warning: 'WebView' is deprecated: first deprecated in macOS 10.14 - No longer supported; please adopt WKWebView. [-Wdeprecated-declarations]
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/System/Library/Frameworks/WebKit.framework/Headers/WebView.h:119:12: note: 'WebView' has been explicitly marked deprecated here
# github.com/wally/vendor/github.com/fdidron/webview
In file included from vendor/github.com/fdidron/webview/webview.go:27:
vendor/github.com/fdidron/webview/webview.h:84:3: warning: 'WebView' is deprecated: first deprecated in macOS 10.14 - No longer supported; please adopt WKWebView. [-Wdeprecated-declarations]
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/System/Library/Frameworks/WebKit.framework/Headers/WebView.h:119:12: note: 'WebView' has been explicitly marked deprecated here
In file included from vendor/github.com/fdidron/webview/webview.go:27:
vendor/github.com/fdidron/webview/webview.h:1691:17: warning: 'WebView' is deprecated: first deprecated in macOS 10.14 - No longer supported; please adopt WKWebView. [-Wdeprecated-declarations]
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/System/Library/Frameworks/WebKit.framework/Headers/WebView.h:119:12: note: 'WebView' has been explicitly marked deprecated here
# github.com/wally/zsa/wally
./ui_dev.go:13:11: undefined: wally.NewState
```

owiefawefoeiwf, 2020-08-03:
on master, i get

```
$ go build cli/main.go -o wally-cli
can't load package: package cli/main.go is not in GOROOT (/Users/username/homebrew/Cellar/go/1.14.6/libexec/src/cli/main.go)
can't load package: package -o is not in GOROOT (/Users/username/homebrew/Cellar/go/1.14.6/libexec/src/-o)
can't load package: package wally-cli is not in GOROOT (/Users/username/homebrew/Cellar/go/1.14.6/libexec/src/wally-cli)
```

psionic-k, 2020-12-04:
There's a nix package for the CLI tool, and it supports build on darwin

https://github.com/NixOS/nixpkgs/commit/b79b4c0eb09932669a731410bb1df773cd03336e#diff-b7566fd6e5332659593394be61161a5a3d7a1c3846a4c5e7f422dccd52a9691f

Suggest adopting nix over homebrew for better reproducible builds.  Just `nix run nixpkgs.wally-cli -c 'wally-cli path/to/my/rom.hex'` once you have nix installed