zsa/wally #40: add CLI to the path

danielo515, 2020-01-24:
When you install wally the CLI tool is not added to the path.
Will it be possible to add that step?
Thanks and regards

DiscoRiver, 2020-01-24:
What OS are you using? And exactly how are you installing?

If, by chance, you're using Linux, you're responsible for the binary's location once it's downloaded, so you should move it somewhere like `/usr/local/bin` (which is already in your $PATH), according to FHS: http://pathname.com/fhs/

danielo515, 2020-01-25:
Sorry for the lack of detail. I'm using OSX Catalina, and I'm downloading the installer from the official page of the project.

Regards

DiscoRiver, 2020-01-28:
No worries! Sorry for the delay, it's been a crazy few days. 

I've got a couple of ideas for this, so I'm going to test those and see which one works best. I'll keep you posted.

DiscoRiver, 2020-01-28:
Okay so I've updated the dmg.json to include the /usr/local/bin directory;

```
{
  "title": "Wally",
  "icon": "Wally.app/Contents/Resources/wally.icns",
  "background": "background.png",
  "contents": [
    { "x": 50, "y": 50, "type": "file", "path": "wally-cli" },
    { "x": 50, "y": 242, "type": "file", "path": "Wally.app" },
    { "x": 242, "y": 242, "type": "link", "path": "/Applications" },
    { "x": 242, "y": 50, "type": "link", "path": "/usr/local/bin" }

  ],
  "window": {
    "size": {
      "width": 400,
      "height": 350
    }
  }
}
```

 and added a background image for user-simplicity. It looks like this now;

[image]

This seems to be better than having users place the wally-cli binary directly into the applications directory.

I don't have access to push any new code to this repo, but hopefully it helps (and I'd be happy to contribute more if they let me).

danielo515, 2020-01-28:
That looks fantastic!

DiscoRiver, 2020-02-19:
@danielo515 - I've been doing internal company work for so long I forgot how to open source. 

I'm going to create a PR tomorrow and try to get this in.

danielo515, 2020-02-20:
Hahaha, no worries. I already have the full path to wally cli on my bash history, so it doesn't require that much typing 😂

fdidron, 2020-06-15:
@DiscoRiver @danielo515 Thanks for this, I'm cooking a new version of Wally and will make sure it includes this

closed 2020-06-15 by fdidron