zsa/wally #42: Windows assertion failure

mykter, 2020-02-18:
Whilst flashing an image (from Oryx) to an ergodox EZ - that seems to have succeeded - this error appeared:
[image]

`File: os/poll_windows.c Line 145 Expression fd != NULL`

Windows Version 1.1.4

I pressed Ignore, and found this in the log (which incidentally has a very weird selection and copy&paste behaviour):
```
09:17:17 info Application UI / State initialized
09:17:18 info Probing compatible usb devices
09:17:18 info Found 1 compatible device(s)
09:17:25 info Starting Teensy Flash
09:17:25 info Waiting for a DFU capable device
...
09:17:38 info Sent 0 bytes out of 32256
...
09:17:39 info  Sent 32000 bytes out of 32256
09:17:39 info Sent 32128 bytes out of 32256
...
09:17:39 info Sending the reboot command
09:17:39  error Error while rebooting device: libusb: i/o error [code -1]
```

The keyboard seemed to behave fine without me manually rebooting it, and the changes in the new firmware were present.

DiscoRiver, 2020-02-23:
- Has this happened more than once? 

[This](https://github.com/libusb/libusb/blob/6816bb40a3bf3e80eb7f86f4dd368f8be76667c5/libusb/os/poll_windows.c#L373) libusb assertion is failing and terminating the application. 

Basically, it seems libusb failed to find the file descriptor. I suspect the device was disconnected in some manner (not blaming you), or there is a bug in libusb causing it to grab a bad descriptor. 

I have limited knowledge of MinGW and libusb, but if this becomes a bigger issue than just a single occurence, I can look into it.

mykter, 2020-02-26:
No, just the once. I will report back if it happens again.

closed 2020-02-26 by mykter

DroogaMcGee, 2020-03-29:
This same thing is happening to me. I have tried flashing my Ergodox EZ three times now with the default Windows layout and consistently get the Line 145 assertion failed window.
[image]

DrewOnFire, 2020-03-29:
Looks like it's a flag setting in Chrome that was causing it for me. 
Go to chrome://flags/ and set the following flag to Disabled.
[image]

After this, open Windows Device Manager and delete the Ergodox EZ device in the USB devices section. Then unplug and replug the keyboard. Should fix things.

CodingDive, 2020-07-21:
I get the same error every time I try to flash. Setting the flag, unplugging and restarting my machine did not help

CodingDive, 2020-07-21:
Related to https://github.com/zsa/wally/issues/25. When I plugged the keyboard straight into my PC instead of an extra USB hub, it worked on first attempt. :)