zsa/wally #46: Numpad Issues

Rostiger, 2020-04-04:
This is more a word of advice than an issue. I'm sorry if this isn't the proper place to put it.

In my [layout](https://configure.ergodox-ez.com/planck-ez/layouts/Qa0am/latest) I created a layer (called Extend) with a numpad and mousemode.

In the dropdown for digits there are the options _Numpad 1_ and _Digits 1_. When choosing _Numpad 1_ the keys won't actually work. My suspicion is that this is caused by the number block not actually being enabled (this would need to be set to another key like many keyboard with actual, physical numpads have), but I haven't confirmed it.

So instead of flashing your keyboard like an idiot five times (like me), just set your numpad keys to digits and you'll be good to go.

Oh and also Plancks ARE THE BESTESTTTT!11 <3 <3 <3

DiscoRiver, 2020-04-04:
The NUMPAD keys function exactly how they are supposed to, meaning that NUMLOCK must be enabled. You can use NUMPAD keys but I would suggest adding the NUMLOCK key to your layer.

Some uses may require the specific use of NUMPAD keys so changing them to digits is not always a viable solution.

Rostiger, 2020-04-04:
Right, I should have said, it didn't work as **I** expected. Of course it works just like it's supposed to, thanks for the clarification and confirmation!

DiscoRiver, 2020-04-04:
It's all good! It can't hurt to have this logged for anyone running into similar issues.

fdidron, 2020-06-15:
@Rostiger glad you got this sorted ! If you run into any other issue, shoot us an email contact@ergodoz-ez.com

closed 2020-06-15 by fdidron