zsa/wally #52: GUI client hands on "Please don't unplug your board." screen on MacOS in V1.1.3

mdbernard, 2020-06-24:


mdbernard, 2020-06-24:
Does not always get stuck, but never seems to move past this screen when it does (it usually moves past this screen after just a few seconds).

mdbernard, 2020-06-24:
Board seems to work just fine with updated layout if Wally is closed after a minute.

arcaartem, 2020-06-24:
When it gets stuck, do you see a little red notification towards the bottom rightish corner of the app? If so, might worth checking the logs to see if you're facing a similar issue to this one: https://github.com/zsa/wally/issues/3

fdidron, 2020-06-25:
@mdbernard I think @arcaartem is right this is similar to #3 
I'm working on a fix for that, but until then, it is totally fine to unplug / plug the board at this stage