zsa/wally #53: Bad checksum for wally on homebrew

xaocon, 2020-06-26:
Hi,

I tried to use homebrew to install on 2 different OSX systems and deleted the cached version and tried again. Each time I get the same SHA for the download which is different from the one expected in the formula. I am fully up to date on brew. I'm not sure this is the right place for this but I was concerned there was a small chance that this was a security issue so wanted to inform as quick as possible.

```
==> Downloading https://github.com/zsa/wally/releases/download/1.1.3-osx/wally-osx-1.1.3.dmg
==> Downloading from https://github-production-release-asset-2e65be.s3.amazonaws.com/186514878/8d68b380-b5a2-11ea-9373-dc9491d2ec78?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20200626%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20200626T160604
######################################################################## 100.0%
==> Verifying SHA-256 checksum for Cask 'zsa-wally'.
==> Note: Running `brew update` may fix SHA-256 checksum errors.
Error: Checksum for Cask 'zsa-wally' does not match.
Expected: 3e6327f23743e3163a7bed4130f73830ecaeea400098003354ec8c63e0010095
  Actual: 9dc788e16b0dd3336748867c1b5b1c06f2c55010d5b6a037cd5cbf22c67b61ef
```

xaocon, 2020-06-26:
Looks like it's probably just related to https://github.com/zsa/wally/issues/27#issuecomment-648159483

I'll leave this open since you'll probably still want to fix the cask.

fdidron, 2020-06-27:
@xaocon Ohhhh good point, thanks for reporting, I'll update the cask.

fdidron, 2020-07-31:
Hi @xaocon waited to release v2 of Wally, the cask should be updated soon:
https://github.com/Homebrew/homebrew-cask-drivers/pull/1677

Thanks again

closed 2020-07-31 by fdidron