zsa/wally #54: handle_events: error: libusb: interrupted [code -10]

turion, 2020-07-04:
I tried flashing my brand new ergodox with firmware downloaded from https://configure.ergodox-ez.com/ergodox-ez/layouts/ABWd0/latest/0:

```
$ wally ergodox_ez_turion-neo2_ABWd0_Jv40Z.hex 
⠦ Press the reset button of your keyboard.2020/07/04 17:17:37 handle_events: error: libusb: interrupted [code -10]
⠼ Press the reset button of your keyboard.
⠼ Press the reset button of your keyboard.
⠴ Press the reset button of your keyboard.2020/07/04 17:18:32 handle_events: error: libusb: interrupted [code -10]
⠹ Press the reset button of your keyboard.2020/07/04 17:18:32 handle_events: error: libusb: interrupted [code -10]
⠋ Press the reset button of your keyboard.2020/07/04 17:18:35 handle_events: error: libusb: interrupted [code -10]
⠦ Press the reset button of your keyboard.2020/07/04 17:18:39 handle_events: error: libusb: interrupted [code -10]
⠙ Press the reset button of your keyboard.2020/07/04 17:18:41 handle_events: error: libusb: interrupted [code -10]
⠹ Press the reset button of your keyboard.2020/07/04 17:18:51 handle_events: error: libusb: interrupted [code -10]
⠧ Press the reset button of your keyboard.2020/07/04 17:18:53 handle_events: error: libusb: interrupted [code -10]
⠙ Press the reset button of your keyboard.2020/07/04 17:19:11 handle_events: error: libusb: interrupted [code -10]
⠸ Press the reset button of your keyboard.^C
```
I pressed the reset button a few times, but it didn't help.

turion, 2020-07-04:
Needless to say, my keyboard doesn't work anymore now.

codygman, 2020-10-02:
I also got this error using the package from nixpkgs referenced in #19. Luckily for some reason my keyboard is still working (maybe I Didn't hold reset long enough).

kush99993s, 2021-04-13:
I am facing a similar issue. Though I am able to use the keyboard. but not able to update layout