zsa/wally #55: Multiple issues with Info.plist on macOS

core-code, 2020-07-11:
the app has no version numbers. all Mac apps are supposed to have version information so that users can look at the version number in the Finder and so that the system can determine the newest version when more than one is installed

1.) the CFBundleVersion is missing in the Info.plist file
2.) the CFShortVersionString is missing in the Info.plist file
3.) the CFBundleGetInfoString has been deprecated for more than 10 years and should be removed

fdidron, 2020-07-31:
@core-code Thanks for reporting this, it is now fixed on the new OS X Release https://github.com/zsa/wally/releases/tag/2.0.0-osx

closed 2020-07-31 by fdidron

core-code, 2020-07-31:
thanks! i can confirm this is resolved in version 2.0.0