zsa/wally #56: Webkit and X window system issue on Ubuntu 20.04

flyinggrizzly, 2020-07-15:
I've just followed the Linux setup instructions, and tried to flash a new firmware, and hit this:

The system is a fresh Ubuntu 20.04 install. I'm not sure what else you need, but I'm happy to provide more system information!

```bash
% ./wally

(WebKitWebProcess:249936): Gdk-ERROR **: 19:13:55.911: The program 'WebKitWebProcess' received an X Window System error.
This probably reflects a bug in the program.
The error was 'BadValue (integer parameter out of range for operation)'.
  (Details: serial 211 error_code 2 request_code 152 (GLX) minor_code 34)
  (Note to programmers: normally, X errors are reported asynchronously;
   that is, you will receive the error a while after causing it.
   To debug your program, run it with the GDK_SYNCHRONIZE environment
   variable to change this behavior. You can then get a meaningful
   backtrace from your debugger if you break on the gdk_x_error() function.)
```
This only appears if I start the Wally gui from the terminal--if I double-click the icon in the file browser, the errors don't appear.

In either case, Wally launches, but just continues to search for flashable devices, finding none.

fdidron, 2020-07-15:
Hi thanks for reporting,

There seems to be 2 issues here, the X window one is a known issue coming from a dependency Wally is using. Once an update lands, I will update Wally.

As for Wally not detecting your boards, can you confirm 2 points :
1. Did you set the udev rules as explained here: https://github.com/zsa/wally/wiki/Linux-install#2-create-a-udev-rule-file
2. Is your board connected directly to your computer, or via a dongle / hub ?

flyinggrizzly, 2020-07-16:
Hi, no worries. I appreciate all the work y'all put in.

> 1. Did you set the udev rules as explained here: zsa/wally/wiki/Linux-install#2-create-a-udev-rule-file

Yes, it was a straight copy-paste. I've got an Ergodox EZ Original (no lights), with plate-mounted switches, if it matters. I did not remove the Planck entries from the rules.

> 2. Is your board connected directly to your computer, or via a dongle / hub ?

Originally via hub, but then when it failed to find the board I made a direct connection, and it still couldn't find the board. I waited a good few minutes too. The board is discoverable on my Mac with Teensy without any problems, so I don't *think* it's the board.

mrtndwrd, 2020-10-21:
Hi, I'm also running into the X Window System error: 

```
./wally

(WebKitWebProcess:97521): Gdk-ERROR **: 16:41:48.682: The program 'WebKitWebProcess' received an X Window System error.
This probably reflects a bug in the program.
The error was 'BadValue (integer parameter out of range for operation)'.
  (Details: serial 209 error_code 2 request_code 151 (GLX) minor_code 34)
  (Note to programmers: normally, X errors are reported asynchronously;
   that is, you will receive the error a while after causing it.
   To debug your program, run it with the GDK_SYNCHRONIZE environment
   variable to change this behavior. You can then get a meaningful
   backtrace from your debugger if you break on the gdk_x_error() function.)
```

Any way to work around this? Or is there an older Wally version I can also use with newly downloaded layouts? Thanks!

mrtndwrd, 2020-11-27:
I just found out that I didn't have `libwebkit2gtk-4.0-37` installed. I'm reasonably sure I fixed this problem by running `sudo apt install libwebkit2gtk-4.0-37` (but I also re-downloaded `wally`, so I'm not sure if anything else changed.