zsa/wally #63: CLI doesn't seem to be available anymore

synaptiko, 2020-07-31:
I was about to update https://aur.archlinux.org/packages/zsa-wally/ but I do not see `wally-cli` binary available anymore and `main/cli.go` fail which is mentioned in [README.md](https://github.com/zsa/wally#compile-the-cli) doesn't seem to be present.

Was it removed? Is there any replacement for it? I personally was using it quite often when fine-tuning my mappings for ErgodoxEZ in QMK.

fdidron, 2020-07-31:
@synaptiko Wally 2.0.0 GUi was a big rewrite, and I'm currently working on moving some of those changes to the cli. I also might put the cli in its own repo. I'm keeping this open for now and will let you know once it's ready.

fdidron, 2020-07-31:
And I use the cli too, so you can bet it's here to stay :)

mejibyte, 2020-08-26:
Hi there, any updates around bringing the CLI back? Thanks!

chrisfair, 2020-08-28:
Hey fdidron I see what you are doing here I think I might be able to help with this issue how do I go about adopting it?   I have 6 kids and a mainline job of course but if nobody else gets on it I think that I can.   The first thing we can do is move the wally dir into it's own repo.   At that point I can write the cli interface pretty easily I think.   If wally is in its own repo then a small addition to the go.mod would allow you to import it like anything else.   Then both would be dependent on the same code.   I would have fun with it if you want help?   I have only written a couple of things in Go (only one professionally) but am currently mucking around with a wails app for funsies already.   BTW I do have a vested interest in this as I currently own an ergodox ez both at home and and at work (although weirdly both home and work are the same for me at present lol).

mihaiolteanu, 2020-09-14:
Well, this is annoying.

fdidron, 2020-09-25:
Hello, I moved the cli to its own repo here: https://github.com/zsa/wally-cli and added support for the Moonlander as well.

@chrisfair thanks for the offer, if there are things you want to add to the cli or improve, we can move the discussion to the new repo.

closed 2020-09-25 by fdidron