zsa/wally #64: 2.0.0 Will not run on OS X Catalina

curtism99, 2020-07-31:
I downloaded the new Wally 2.0, copied it to my local Applications folder, and double clicked to launch.

I receive a popup error that the title is "Wally cannot be opened because of a problem."

The body text of the error is: "Check with the developer to make sure Wally works with this version of macOS. You may need to reinstall the application. Be sure to install any available updates for the application and macOS.

Click Report to see more detailed information and send a report to Apple."

Would appreciate help understanding what's wrong here.
Thanks!

fdidron, 2020-07-31:
Hi @curtism99 

There should be a button to get the full log, could you paste the log in here ? thanks

curtism99, 2020-07-31:
@fdidron 
Here is the full log in the error report upon opening.
```

Process:               Wally [39050]
Path:                  /Volumes/*/Wally.app/Contents/MacOS/Wally
Identifier:            com.zsa.wally
Version:               2.0.0
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           Wally [39050]
User ID:               501

Date/Time:             2020-07-31 06:26:05.066 -0500
OS Version:            Mac OS X 10.15.6 (19G73)
Report Version:        12
Bridge OS Version:     4.6 (17P6065)
Anonymous UUID:        83CD0968-7217-4025-9390-C5B4FB86A6D5

Sleep/Wake UUID:       2C945817-A3B5-44A7-818F-AF8D50A5E972

Time Awake Since Boot: 230000 seconds
Time Since Wake:       4100 seconds

System Integrity Protection: enabled

Crashed Thread:        0

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    DYLD, [0x1] Library missing

Application Specific Information:
dyld: launch, loading dependent libraries

Dyld Error Message:
  Library not loaded: @executable_path/../libs/libusb-1.0.0.dylib
  Referenced from: /Volumes/*/Wally.app/Contents/MacOS/Wally
  Reason: no suitable image found.  Did find:
	file system relative paths not allowed in hardened programs

Binary Images:
         0x4000000 -          0x4a13fff +com.zsa.wally (2.0.0) <DCC43A8E-ACE4-3A6D-83AC-D3E4E917D47B> /Volumes/*/Wally.app/Contents/MacOS/Wally
         0xf429000 -          0xf4baf47  dyld (750.6) <34A11073-9E4C-38C3-9293-7D566ABAE8B6> /usr/lib/dyld

Model: MacBookPro15,1, BootROM 1037.147.1.0.0 (iBridge: 17.16.16065.0.0,0), 8 processors, 8-Core Intel Core i9, 2.3 GHz, 16 GB, SMC 
Graphics: kHW_IntelUHDGraphics630Item, Intel UHD Graphics 630, spdisplays_builtin
Graphics: kHW_AMDRadeonPro560XItem, Radeon Pro 560X, spdisplays_pcie_device, 4 GB
Memory Module: BANK 0/ChannelA-DIMM0, 8 GB, DDR4, 2400 MHz, Micron, 8ATF1G64HZ-2G6E1
Memory Module: BANK 2/ChannelB-DIMM0, 8 GB, DDR4, 2400 MHz, Micron, 8ATF1G64HZ-2G6E1
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x7BF), wl0: Feb 28 2020 15:24:56 version 9.30.357.35.32.5.47 FWID 01-9ce4adf3
Bluetooth: Version 7.0.6f7, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB 3.1 Bus
USB Device: USB2.0 HUB
USB Device: USB2.0 HUB
USB Device: USB2.0 Hub
USB Device: USB Receiver
USB Device: USB 10/100/1000 LAN
USB Device: Razer BlackWidow Ultimate 2016
USB Device: My Passport 2626
USB Device: Apple T2 Bus
USB Device: Touch Bar Backlight
USB Device: Touch Bar Display
USB Device: Apple Internal Keyboard / Trackpad
USB Device: Headset
USB Device: Ambient Light Sensor
USB Device: FaceTime HD Camera (Built-in)
USB Device: Apple T2 Controller
Thunderbolt Bus: MacBook Pro, Apple Inc., 47.4
Thunderbolt Bus: MacBook Pro, Apple Inc., 47.4

```

fdidron, 2020-07-31:
@curtism99 Super helpful, I'm working on a fix right now

fdidron, 2020-07-31:
Allright @curtism99 I updated the dmg, can you download it again and let me know if it works for you ?
https://github.com/zsa/wally/releases/tag/2.0.0-osx

curtism99, 2020-07-31:
@fdidron 
Yes! I am able to open the app. Though, I have another problem. i've run into this before with this keyboard, so I actually have never gotten it working properly. I had just a random idea to try it again today.

So this is probably another issue, or need help with customer support. I can now open Wally 2.0 but it just gets frozen on the "Looking..." stage. IT does not find any compatible device(s). I've run the Package Contents terminal app to start the GUI, and it says that there were "handle-events: error: libusb: interrupted [code -10]" as it tries over and over again.

I've also reset the keyboard, tried unplugging and replugging it. I'm not sure what to do next with this to get the keyboard recognized and to install it correctly.

Thank you for your responsiveness!

fdidron, 2020-07-31:
@curtism99 Are you able to plug your Ergodox directly to the Mac (without using the lighting to usb adapter) and try if Wally's able to detect your keyboard.

curtism99, 2020-07-31:
@fdidron No, I unfortunately cannot. My model of MacBook Pro doesn't have any typical USB 3 ports. It only has 4 USB-C/Lightning ports on it for charging, usb, hdmi display output, etc. and adapters are needed for most things.

I did find a Reddit thread mentioning this, and they recommended the right kind of cable to go directly from the Ergodox keyboard to USB-C [for about $10 on Amazon](https://www.amazon.com/gp/product/B07JY9FK2L/?th=1). So it seems like that may be what I need to get in order to use it directly.

If I've got a developer's ear here, it would be an interesting bug or new feature to work on perhaps? I'm using [this Amazon Basics adapter](https://www.amazon.com/gp/product/B01GGKYYT0/ref=ppx_yo_dt_b_asin_title_o03_s00?ie=UTF8&psc=1) which seems to do the job for everything else I've encountered or had to use so far. Regardless, thank you for sparking the idea of finding the right cable to plug directly into the USB-C without the adapter. Seems like that may be what I need to do.
Thanks!

fdidron, 2020-08-02:
@curtism99 Got it, I think a good alternative would be to flash using [Teensy loader](https://www.pjrc.com/teensy/loader_mac.html) . They seem to interact with USB in a way that avoids most issues with hubs / dongles. It's a safe option, it's the tool developed by the Ergodox's MCU manufacturer.

I'm looking into refactoring Wally's usb code for Mac and Windows to tap into native APIs and avoid such issues. We are using a 3rd party lib calle libusb right now, and I think that's the culprit.