zsa/wally #67: Unable to flash ZSA Planck on Windows 10 1903

ayisakov, 2020-08-12:
I cannot flash my Planck using Wally 2.0.0 on Windows 10 1903 64-bit.

Wally recognizes that a Planck is plugged in on startup. I load the firmware file, and Wally tells me to press the reset button on my Planck. Nothing happens when I do so. The following lines loop forever in the console:

```
warning OpenDevices: libusb: not found [code -5]
   info Waiting for a DFU capable device
```

I usually have it connected via a USB 3.0 hub and A/B switch. I connected it directly to the computer's USB port, and the issue persists. I swapped the cable for the original cable that came with the Planck, and the issue persists.

I was able to flash the board using Wally v1.1.1 on Ubuntu 18.04.

ayisakov, 2020-08-20:
[This](https://github.com/Ralim/ts100/issues/298#issuecomment-659215942) may be my issue. I do have Thrustmaster USB devices connected to that machine, and the Guillemot driver immediately replaces any other driver I install for the bootloader, even if I tell Windows to remove the driver.

fdidron, 2020-11-25:
Hi Ayisakov

There's a new version of Wally for Windows here https://github.com/zsa/wally-win . Can you try to stop all Guillemot / Thrusthmaster services, Install the new version and give it a try ? Feel free to open a new issue in the new repo if needed.

closed 2020-11-25 by fdidron