zsa/wally #72: Trying to build the CLI for v2 on Linux

steveeJ, 2020-09-29:
I'm trying to update the package definition for [_wally-cli_ on nixpkgs](https://github.com/NixOS/nixpkgs/blob/b5dc07a4b471e6f651af3b13e4bd841b6b3ea6ed/pkgs/development/tools/wally-cli/default.nix) and I'm confused by the state of this repository, so someone please help me out with the following questions :slightly_smiling_face: 

Is there still a CLI in v2?
If so how do I build only the CLI part on Linux?

Thanks :pray:

fdidron, 2020-09-30:
Hi @steveeJ 

I moved the cli to its own repo. You can find it here https://github.com/zsa/wally-cli

Can you give it a try and open an issue there if you need help ?

Thanks

closed 2020-09-30 by fdidron

steveeJ, 2020-10-03:
Thanks @fdidron, I didn't see that repo. Maybe you could add it to the README here?