zsa/wally #81: ChromeOS instruction page at Wiki

draftcode, 2020-10-24:
wally-cli works well with ChromeOS's Linux support. Adding an instruction would help some people. At first, I also didn't realize that I could do that.

# Steps

1. Enable "Linux (Beta)" in Chrome OS settings.
2. Download [wally-cli](https://github.com/zsa/wally-cli/releases) and a firmware file into the Linux folder.
3. `chmod +x wally-cli`
4. `./wally-cli FIRMWARE_FILE.bin`
5. Press the reset button on the keyboard.
6. ChromeOS shows a pop-up saying a new USB device is detected. Make it available to the Linux environment by clicking the popup or go to the settings page and allow it.
7. wally-cli writes the firmware. Keyboard will be rebooted automatically.

---

As far as I can see, the crostini environment already has libusb and it doesn't need a udevd rule or something to interact with the keyboard.

alon91, 2020-11-21:
I love Chrome OS and now use it exclusively, which means I own zero Windows or Mac devices. I've just enabled Linux on my device but am not having luck with the instructions above. Would love to improve the guidance so others who are similarly struggling can figure it out in the future. Here's what I'm seeing, with screenshots:

**1. Enable "Linux (Beta)" in Chrome OS settings. ✅** 
[image]
[image]

**2. Download [wally-cli](https://github.com/zsa/wally-cli/releases) and a firmware file into the Linux folder. ❓** 
* Not really sure I'm downloading the right thing. Your link points to [here](https://github.com/zsa/wally-cli/releases) while ergodox-ez.com links to [here](https://ergodox-ez.com/pages/wally). Are your instructions specifically for the CLI ([command line interface](https://www.geeksforgeeks.org/linux-operating-system-cli-command-line-interface-and-gui-graphic-user-interface/#:~:text=The%20Command%20Line%20Interface%20(CLI,CLI)%20environment%20to%20the%20user.)) version?
* What is a firmware file? Is it the .hex file that is generated when I hit the "Download this layout" button on a page like [this](https://configure.ergodox-ez.com/ergodox-ez/layouts/gYme0/latest/0)?
[image]

**3. chmod +x wally-cli ❓**
Is anything supposed to happen when I do this? I get no response from the system when I do this.

**4. ./wally-cli FIRMWARE_FILE.bin ❓**
I'm getting an error when I try to append the .hex file I downloaded in step 2
[image]

**5. Press the reset button on the keyboard. ❓**
What button is that?

6. ChromeOS shows a pop-up saying a new USB device is detected. Make it available to the Linux environment by clicking the popup or go to the settings page and allow it.
🤷 

7. wally-cli writes the firmware. Keyboard will be rebooted automatically.
🤷 

Would love the community's help here, and happy to draft up painfully verbose instructions for future Linux dummies like myself. Thank you!

xa888s, 2020-11-22:
You spelled cli wrong, should be ./wally-cli not ./wally-clie

jimpriest, 2020-11-23:
When I download wally on Linux I get 'wally'  not 'wally-cli'  

Then you just make it executable via the chmod + x wally 

And finally run it:  ./wally  

And a gui should popup on the screen.   You can then select your downloaded .hex file. 

I don't have a Chromebook to test with but I just flashed my Ergodox this evening and these were the steps using Ubuntu Linux.

random-un, 2020-11-23:
> You spelled cli wrong, should be ./wally-cli not ./wally-clie

🤦 wow thanks for catching that really dumb mistake! Appreciate the suggestions from both you and @jimpriest. I'm making progress here, although continue to run into problems with both `wally-cli` and `wally`:

## wally-cli
When I enter `./wally-cli test.hex` I now see the message:
> ⠦ Press the reset button of your keyboard

The first character with the dots is animated. When I hit my keyboard's reset button the keyboard's Glow LEDs freeze but the dots animation on my screen continues and the keyboard doesn't flash. 

## wally
I'm getting the error
> ./wally: error while loading shared libraries: libwebkit2gtk-4.0.so.37: cannot open shared object file: No such file or directory

## Troubleshooting

Perhaps this is unsuccessful because of details described in step 1 or 2 of [this](https://github.com/zsa/wally/wiki/Linux-install) page? I attempted to follow everything here but there are not specific instructions for Chromium or [Gentoo](https://www.quora.com/What-specific-version-of-Linux-is-Chrome-OS-based-off-of) so I just kind of tried stuff.

Thanks again for any advice here!

draftcode, 2020-11-23:
> When I hit my keyboard's reset button the keyboard's Glow LEDs freeze but the dots animation on my screen continues and the keyboard doesn't flash.

Did you follow this instruction?

> ChromeOS shows a pop-up saying a new USB device is detected. Make it available to the Linux environment by clicking the popup or go to the settings page and allow it.

jimpriest, 2020-11-23:
libwebkit2gtk-4.0.so.37 looks like some kind of gui/web library which must be missing in Chrome Linux.   So wally-cli is the way to go.

fdidron, 2020-11-24:
@draftcode Thanks a lot for the instructions, I'm going to add that to the wiki.

@random-un Here are my answers inline:

> 2. Download wally-cli and a firmware file into the Linux folder. ❓
> Not really sure I'm downloading the right thing. Your link points to here while ergodox-ez.com links to here. Are your instructions specifically for the CLI (command line interface) version?

These instructions are for the cli indeed, so the binary you want to download can be found here: https://github.com/zsa/wally-cli/releases
> What is a firmware file? Is it the .hex file that is generated when I hit the "Download this layout" button on a page like this?

This is correct, Oryx coverts your layout into a firmware file (.hex for the Ergodox, .bin for the Moonlander/Planck) that is ready to flash into your board using Wally.

> 3. chmod +x wally-cli ❓
> Is anything supposed to happen when I do this? I get no response from the system when I do this.

If the command worked successfully there shouldn't be any response.

> 4. ./wally-cli FIRMWARE_FILE.bin ❓
> I'm getting an error when I try to append the .hex file I downloaded in step 2

Looks like you misspelled the command :) thanks for spotting @xa888s 

> 5. Press the reset button on the keyboard. ✅
> I believe this is the button in the upper-right corner of the keyboard that you have to press with a paper clip.

This is correct.

> 6. ChromeOS shows a pop-up saying a new USB device is detected. Make it available to the Linux environment by clicking the popup or go to the settings page and allow it.
> 🤷

You didn't get the pop up to show up after pressing the keyboard's reset button ?

> 7. wally-cli writes the firmware. Keyboard will be rebooted automatically.
> 🤷

This is what's supposed to happen once you passed step 6

> I'm getting the error
> ./wally: error while loading shared libraries: libwebkit2gtk-4.0.so.37: cannot open shared object file: No such file or directory

Let's get the CLI thing sorted first then we can look into that later but as @jimpriest you are missing a dependency Wally's GUI is relying on.

draftcode, 2020-11-24:
I feel like it'd be difficult to make the GUI version work on the ChromeOS's Linux container. I wouldn't say it's impossible, but harder than installing the CUI version.

Technically, what Wally does should be possible with WebUSB. This means that Oryx can directly interact with the keyboards and write the firmware. Considering that ZSA created a training program with that (https://blog.zsa.io/1912-live-training/), I think they have some knowledge to make it so. Not sure if it's on their roadmap, but that will be a better UX overall.

fdidron, 2020-11-24:
@draftcode I developed the webusb implementation of QMK and Oryx. Webusb forbids to claim hid devices, and unfortunately the Ergodox MCU only exposes an hid endpoint when in reset mode.

I agree with you, being able to flash directly from Oryx would have been neat especially for Chrome OS.

Do you know if it is feasible to run GTK apps on the Chrome’s OS container ?

random-un, 2020-11-24:
I'll continue to thank you for all this advice! And also really appreciate these further conversations on potential improvements of flashing in Chrome OS. Quick updates:

@fdidron asked:
> You didn't get the pop up to show up after pressing the keyboard's reset button ?

and @draftcode originally suggested:
> ChromeOS shows a pop-up saying a new USB device is detected. Make it available to the Linux environment by clicking the popup or go to the settings page and allow it.

Unfortunately clicking the reset / flash button on my keyboard doesn't trigger a pop-up on my machine. The keyboard's LED animation stops and the keyboard no longer sends keystrokes so I know it's waiting to be flashed (I have to unplug / re-plug to get it working again).

In Chrome OS settings I went to **Linux (Beta)** --> **USB preferences** but the page doesn't list any USB devices.

[image]

I've been emailing with Erez from ZSA for a bit and he mentioned that they may be able to create an Android app one day but sounds like nothing is imminent. Would be awesome to find a better solution for Chrome OS in the meanwhile!

jimpriest, 2020-11-24:
Did you do the steps mentions in Linux install for adding udev rules, etc?   I just did this last night on my other linux box and had the same issue - ran Wally and hit reset button on keyboard but nothing happened. Then I remembered I needed to do this additional config.  I'm assuming Linux on ChromeOS would have the same requirements. 

https://github.com/zsa/wally/wiki/Linux-install

draftcode, 2020-11-24:
> Webusb forbids to claim hid devices, and unfortunately the Ergodox MCU only exposes an hid endpoint when in reset mode.

Ah, I see. I assumed that it'd be same as Moonlander. In that case, I don't think this instruction works for @random-un, as they seems to have Ergodox. Chrome OS apparently won't let us pass USB HID devices to the Linux sandbox as well. This instruction works only for the keyboards whose reset switch makes it switch to the bootloader mode.

> Do you know if it is feasible to run GTK apps on the Chrome’s OS container?

I found that some people ran GIMP on Chrome OS (searching "crostini GIMP"), so I think it's possible. But the problem is that it wouldn't be able to grab the USB HID devices as-is, so it won't work for the same reason as the CLI version.

random-un, 2020-11-25:
One final comment from me for now: I got further confirmation on other threads / investigations that the Ergodox EZ can't be flashed with Wally on Chrome OS's Linux virtual machine. As mentioned in this thread, it has to do with how forwarding USB HID devices to the Linux VM is currently disallowed by policy on the operating system.

I'd love to see a solution for this, as for now I have to use a very old Macbook as a dedicated flashing laptop. At least I'm not fully blocked here! I've updated my original reply above with the TLDR -- thanks again for all your help, and let me know if there's anything else I can contribute here.

fdidron, 2020-12-06:
@draftcode Here is the wiki page: https://github.com/zsa/wally/wiki/Wally-on-ChromeOS

closed 2020-12-06 by fdidron