zsa/wally #89: Moonlander keyboard bricked after using Wally v2.1.0

lionelbroche, 2020-12-01:
Hi,
I just tried flashing my moonlander keyboard using Wally. I got a .hex file from the Oryx website and tried flashing it. Wally did not accept .hex file so I renamed it .bin and started the flashing process. It completed successfully but the keyboard is now unresponsive. 
Is there a factory reset capability on the keyboard, and what went wrong with the process?

lionelbroche, 2020-12-02:
As an aside, it is not straightforward to find some documentation about recovery processes. If that can help, I managed to reflash the keyboard after pressing the reset switch between the moment the keyboard is powered and the start of the flashing process. This makes the keyboard discoverable.

lionelbroche, 2020-12-02:
ok, the problem is not really with Wally but with the ergodox-ez.com website. When downloading a layout you have to be logged in to get a bin file, otherwise it gives a hex file. I assumed thet were equivalent, but they are definitely not. Not sure if this is something to mention here then.

endafarrell, 2021-03-03:
I have this problem too. On a Mac, wally just seems to hang. `wally-cli` does work with the same `.bin` file. I can supply more details if you need them.