zsa/wally #96: Consider a fully native Go UI

mvdan, 2021-01-05:
I realise this is a big suggestion rather than a bug report, though it is one I thought would still be useful :)

I've been using Wally for a few months since buying a Moonlander, and it's pretty neat. Thanks for maintaining it! I only recently noticed that the UI is actually a web app, which is a bit unfortunate. At least on my Linux box, that pulls in webkit2gtk, which weighs over a hundred megabytes. Having to start a proper web browser with the app also makes it start quite slowly; when I open wally, it takes half a second for a blank window to show up, and another half a second for the actual UI to load up. All in all, a full second to load a pretty simple landing screen.

Luckily, the portable GUI space in pure Go has advanced quite a lot in the past few years. There were previous attempts like x/exp/shiny, but none took off as much as recent projects like Gio - you can see [some examples here](https://gioui.org/#programs-using-gio), in particular the Tailscale app.

I realise portability to Android/iOS is perhaps a non-factor for Wally, but having a single Go program which can show the same UI on Linux, Mac, and Windows seems valuable. Plus it would not require a browser as a dependency, and it would start up faster :)

whereswaldon, 2021-01-05:
I think Gio would be a solid fit for this too! There are some simple in-browser demos of my gio libraries available [here](https://whereswaldon.github.io/gio-extras-index/) to suggest what it can handle, but for a better idea of what's possible, you can look at [gio-example](https://git.sr.ht/~eliasnaur/gio-example) and the other programs that @mvdan linked above.

fdidron, 2021-01-06:
@mvdan I agree with you. When I started working on Wally 2 years ago the promise of having a Cross Platform UI was appealing, I wanted to avoid electron, and since webkit2gtk was already installed on the most popular distros (Ubuntu and its derivatives) it seemed like a good compromise. Also I really like the Go Programming language :)

Now 2 years have passed, and I'm not too thrilled about cross platform anymore. It makes maintaining / publishing way too hard. After reading a bit on how to write native desktop apps, I realized it was not so complex after all.

As a result, I'm currently rewriting Wally for each platform using Native Frameworks:

- The Windows version is already out using C# / WPF https://github.com/zsa/wally-win 

- I'm finalizing the OS X Version Using Swift / UIKit, USB code is done, working on the UI.

Now for Linux, I'm thinking working on a native GTK version using Vala or C, this would remove the webkit2gtk dependency.

fdidron, 2021-01-06:
Also one thing to note, I like the idea of open sourcing a native version for each platform, because while learning how to develop Win / OS X native apps, I realized that there's not so many native open source project doing USB Communication out there. I hope this can benefit someone looking into talking to a USB Device using native frameworks.

mvdan, 2021-01-06:
Gotcha, sounds good. I would personally still have gone for a single portable app, but you've clearly already made progress :)