zsa/wally #105: "Unable to claim interface" whilst training

sgmmahon, 2021-02-15:
Apologies if this issue has already been addressed elsewhere, I could not find a solution when looking through the issues archive.

I am trying to access [Training](https://configure.ergodox-ez.com/train) for my Moonlander keyboard. However when I try to connect my keyboard I get the following error:

[image]

This error has only occurred today. Last week my keyboard was connecting fine.

Any suggestions as to what may be going wrong?

Thanks!

sgmmahon, 2021-02-15:
Restarted computer and issue was resolved.

sgmmahon, 2021-02-17:
Apologies, reopened the issue as this error has reoccured several more times over the last few days.

arcturus140, 2021-02-22:
I am running into the same problem. Restarting the OS did not make a difference. 

- using Planck EZ Default Layout (v18.0) is working without issues
- cloning Planck EZ Default Layout (v20.0) does not work: "No compatible devices found"

my setup:

- Ergodox Planck EZ glow
- Linux Ubuntu 20.04
- libusb-dev is already the newest version (2:0.1.12-32).
- Google Chrome Version 88.0.4324.182 (Official Build) (64-bit)
- Wally v2.1.0

fdidron, 2021-02-22:
Hello,  @sgmmahon can you let me know which OS/Browser versions you are using ?

fdidron, 2021-02-22:
Hi @arcturus140,

Did you setup the udev rules as mentionned in the wiki page: https://github.com/zsa/wally/wiki/Live-training-on-Linux

arcturus140, 2021-02-23:
@fdidron it is now working. I added the udev rule which was missing in my configuration and restarted the machine. Also, for anyone else the reboot was necessary. At first I tried to log-out and in as User and reconnect the keyboard but this did not work. 

thank you very much for your help.