zsa/wally #109: Mac / Moonlander - Waiting for a DFU capable device

NLKok, 2021-02-25:
I am unable to flash my Moonlander keyboard on Mac. Last time I successfuly flashed my Moonlander was about a month ago.  When I open my latest layout in the [configurator](https://configure.ergodox-ez.com/), I also see the firmware changed from `v19.0` to `v20.0`. Maybe that causes the problem.

When open Wally it detects my Moonlander, then I load my new firmware and the moment I press the reset button, my Moonlander got stuck in reset mode and Wally isn't detecting my keyboard. After taht I have to disconnect the keyboard to reboot it.

[image]

Running MacOS Big Sur 11.1 and Wally 2.10.

willhopkins, 2021-02-26:
I just received a Moonlander today and tried to flash the firmware on Linux (KDE Neon 5.21) and am having the exact same problem and error. I downloaded the binary and wally today for the first time. Happy to provide logs or other details on request but every symptom is identical to the ones described here.

LiBa001, 2021-02-26:
Same goes for me on Ubuntu 18.04. I'm already having this problem for a while now and I'm occasionally checking if it's fixed. Which it still isn't. I guess it's since V2.1.0 (October 2020) that this problem occurs and it probably will continue to occur until the next release.

LiBa001, 2021-03-07:
Nevermind, I was just insanely stupid the whole damn time and pressed the Oryx button instead of the reset button.

***facepalm***

NLKok, 2021-03-15:
Don't know what changed. But I just tried downloading my latest firmware and now it all works. Same Wally version, same firmware, no changes, same procedure... Maybe Oryx change something on there side. Since it works I am closing this issue.

closed 2021-03-15 by NLKok