zsa/wally #110: Does Wally support .hex?

aronchick, 2021-03-06:
Hi--

It's not clear if Wally support .hex or not - the ergodox-ez website downloads both bin and hex - bin works fine, but hex does not - and the instructions say ".bin". So I'm _assuming_ .bin is all that works, but does that mean that the ergodox-ez website is useless? Is there any path here?

Thanks!

fdidron, 2021-03-06:
Hi,

Wally supports both format but the .hex file format targets Ergodox Ez keyboards, the .bin file targets The Moonlander and the Planck.

On configure.ergodox-ez.com the Ergodox pages will produce a hex file, the ones for the Moonlander/Planck will produce .bin files.

Which keyboard are you using ?

aronchick, 2021-03-06:
Oh this makes sense - I just didn't realize how to swap Oryx over from (wrongly) detecting I was on an Ergodex (i'm on a moonlander).

Obviously, you can close this because it's my fault - but it might be nice to pop a warning saying "You dragged a .hex but you have a Moonlander attached (or whatever). .hex files are only supported on Ergodex" (etc).

Your call though ... thank you so much for all your work!

adelsz, 2021-03-18:
Got confused by this as well. I renamed the .hex file to .bin and flashed (warning: dont do this!)
That almost bricked the board, it got stuck in flashing mode and stopped responding to anything (reboots/resets/power off didnt help at all). Got really frustrated to brick it less than 5 mins after I got the keyboard.

Finally managed to get the keyboard out of that state after a random combination of resets/power ons and using the wally-cli.