zsa/wally #116: Wally fails to launch in Gnome 40

Archehandoro, 2021-03-26:
Hello,

When launching wally, under fedora 34 beta with gnome 40, I get the following error.

`
SIGABRT: abort
PC=0x7ff5356c4292 m=0 sigcode=18446744073709551610
signal arrived during cgo execution

goroutine 1 [syscall, locked to thread]:
runtime.cgocall(0x7ff860, 0xc00010bbd0, 0x0)
	/usr/lib/go-1.13/src/runtime/cgocall.go:128 +0x5b fp=0xc00010bba0 sp=0xc00010bb68 pc=0x4089bb
github.com/wailsapp/wails/lib/renderer/webview._Cfunc_CgoWebViewCreate(0x20800000276, 0x1937580, 0x19375a0, 0x0, 0x0)
	_cgo_gotypes.go:111 +0x4e fp=0xc00010bbd0 sp=0xc00010bba0 pc=0x7db24e
github.com/wailsapp/wails/lib/renderer/webview.NewWebview(0x904922, 0x5, 0x0, 0x0, 0x276, 0x208, 0x0, 0xc000020040, 0xa, 0xe)
	/home/florian/dev/go/pkg/mod/github.com/wailsapp/wails@v1.8.0/lib/renderer/webview/webview.go:261 +0xfe fp=0xc00010bc30 sp=0xc00010bbd0 pc=0x7dbd5e
github.com/wailsapp/wails/lib/renderer.(*WebView).Initialise(0xc0000fe000, 0xb18dc0, 0xc000100180, 0xb14c00, 0xc000034120, 0xb15a40, 0xc0000bc0a0, 0x203000, 0x203000)
	/home/florian/dev/go/pkg/mod/github.com/wailsapp/wails@v1.8.0/lib/renderer/webview.go:53 +0x358 fp=0xc00010bd28 sp=0xc00010bc30 pc=0x7ddb38
github.com/wailsapp/wails.(*App).start(0xc000100100, 0x0, 0x0)
	/home/florian/dev/go/pkg/mod/github.com/wailsapp/wails@v1.8.0/app.go:115 +0x197 fp=0xc00010be68 sp=0xc00010bd28 pc=0x7e1eb7
github.com/wailsapp/wails.(*App).Run(0xc000100100, 0x8cdf60, 0xc00007a000)
	/home/florian/dev/go/pkg/mod/github.com/wailsapp/wails@v1.8.0/app.go:94 +0xaf fp=0xc00010bee0 sp=0xc00010be68 pc=0x7e1bef
main.main()
	/home/florian/dev/go/src/github.com/zsa/wally/main.go:24 +0x216 fp=0xc00010bf60 sp=0xc00010bee0 pc=0x7fe256
runtime.main()
	/usr/lib/go-1.13/src/runtime/proc.go:203 +0x21e fp=0xc00010bfe0 sp=0xc00010bf60 pc=0x43509e
runtime.goexit()
	/usr/lib/go-1.13/src/runtime/asm_amd64.s:1357 +0x1 fp=0xc00010bfe8 sp=0xc00010bfe0 pc=0x460881

goroutine 19 [syscall]:
os/signal.signal_recv(0x0)
	/usr/lib/go-1.13/src/runtime/sigqueue.go:147 +0x9c
os/signal.loop()
	/usr/lib/go-1.13/src/os/signal/signal_unix.go:23 +0x22
created by os/signal.init.0
	/usr/lib/go-1.13/src/os/signal/signal_unix.go:29 +0x41

rax    0x0
rbx    0x7ff530ec7a80
rcx    0x7ff5356c4292
rdx    0x0
rdi    0x2
rsi    0x7ffe408fcbc0
rbp    0x7ff53584c8f0
rsp    0x7ffe408fcbc0
r8     0x0
r9     0x7ffe408fcbc0
r10    0x8
r11    0x246
r12    0x7ff535a0dba0
r13    0x7ff535a0db70
r14    0x7ff535a0d140
r15    0x0
rip    0x7ff5356c4292
rflags 0x246
cs     0x33
fs     0x0
gs     0x0
`
I guess the requisite of gtk3 conflicts with the newer APIs from gnome 40? Anyone has an idea on how it could I fix it? Thanks!

fdidron, 2021-04-09:
Hi @Archehandoro can you give this release a try ?
https://github.com/zsa/wally/releases/tag/2.1.1-linux

Archehandoro, 2021-04-09:
Hi, I gave it a try and the app launches, but buttons and images don't render, so only see a blank window. In the terminal, this log appears:
```Overriding existing handler for signal 10. Set JSC_SIGNAL_FOR_GC if you want WebKit to use a different signal```
Thanks for your help :)

bart3005, 2021-04-16:
On Fedora 33 I also get a black window:

[image]