zsa/wally #121: libc6 version 2.32 required but Ubuntu 20.04 has 2.31

pjio, 2021-04-09:
Steps to reproduce:

Download wally binary like suggested at: https://github.com/zsa/wally/wiki/Linux-install

```
$ md5sum wally 
92ac2b1c2a021ea4380de4c990089c68  wally
$ dpkg -l | grep libc6:amd64
ii  libc6:amd64                                2.31-0ubuntu9.2                       amd64        GNU C Library: Shared libraries
$ sudo ./wally 
./wally: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.32' not found (required by ./wally)
```
Most recent libc6 for Ubuntu is 2.31
https://packages.ubuntu.com/focal/libc6

Am I missing something or should wally just run?

dylan-sessler, 2021-04-09:
@pjio I was able to flash firmware using the wally-cli with no issues after running into the same issue as you earlier today.

I am running the cli with `./wally-cli` out of a directory where I am storing my keyboard binaries.
https://github.com/zsa/wally-cli/releases

ajoy39, 2021-04-11:
Experiencing the same problem, Ubuntu 20.04 won't install anything > 2.31 so unless you're comfortable compiling from source and replacing the system glibc this build is broken

fdidron, 2021-04-11:
Hi, I upaded the binany here: https://github.com/zsa/wally/releases/tag/2.1.1-linux

Can you give it a try ? @pjio @ajoy39

pjio, 2021-04-12:
**@fdidron** Version 2.1.1 works!
I was able to start the UI and flash a new configuration to my keyboard without errors.

ajoy39, 2021-04-13:
> Hi, I upaded the binany here: https://github.com/zsa/wally/releases/tag/2.1.1-linux
> 
> Can you give it a try ? @pjio @ajoy39

Can confirm as well, 2.1.1 works on Ubuntu 20.04 with GLIBC 2.31